prenticedavid/MCUFRIEND_kbv #176: Elegoo 2.8" TFT Display - White Screen Error

adamfres, 2021-05-02:
I'm using the display with an arduino mega that it is plugged directly into. When running the example programs provided by MCUFRIEND the display stays white. Any help would be greatly appreciated. 
[image]

prenticedavid, 2021-05-02:
Go on.  What do you expect anyone to to see from the photo glare?

Post the output from the Serial Terminal.  Or just say "I get a white screen"

David.

adamfres, 2021-05-02:
Sorry about the unclear photo. When I run graphictest_kbv the serial reads: 

Serial took 0ms to start
ID = 0x9341

The screen stays solid white.

Thanks

prenticedavid, 2021-05-03:
1.  Delete any manually installed library(s)
   2.  Install those libraries properly via IDE Library Manager.
   3.  Plug Shield into Uno. All the male pins should engage with the corresponding female header sockets.
   4.  Run graphictest_kbv from examples
   5.  Copy-paste the Serial Terminal

You seem to be doing things correctly e.g. for a Mega2560.   Please answer (1), (2)
I would expect every example to work 100%.

David.

adamfres, 2021-05-03:
I did step 1 & 2, also re-installed arduino ide. See the photos of how the screen is plugged into the mega. Graphics test still returned: 
Serial took 0ms to start
ID = 0x9341
With the screen remaining white.

[image]
[image]

Thanks

prenticedavid, 2021-05-03:
You seem to be doing everything correctly.
Does the Display Shield have AMS1117 regulator and LV245 buffers or HC245 buffers ?

Have you tried a different Mega2560 or a Uno ?
Do you have a link to the display ?

adamfres, 2021-05-03:
[Elegoo 2.8 inch Touch Screen User Manual（Arduino-English）V1.00.2017.04.12.pdf](https://github.com/prenticedavid/MCUFRIEND_kbv/files/6413572/Elegoo.2.8.inch.Touch.Screen.User.Manual.Arduino-English.V1.00.2017.04.12.pdf)
The specifcation is attached above. I've tried the screen with my friends Arduino Uno and it works correctly. So it suggests the issue is with the mega.

prenticedavid, 2021-05-03:
Does the Display Shield have AMS1117 regulator and LV245 buffers or HC245 buffers ?

Thanks for the PDF.   I am much happier with your eyes reading the part numbers on the pcb than trusting images in the PDF.   

If you installed MCUFRIEND_kbv correctly it should just work on Uno, Mega2560, Due, Nucleo, ...
I suggest that you try a different Mega2560.   Or any board with Arduino Header sockets.

David.

adamfres, 2021-05-03:
[image]
Heres the best photo I can get of the chips.

Thanks

prenticedavid, 2021-05-03:
I am gobsmacked.   Is it not easier to just type LVC245A than posting a high resolution photo?
And saying _Yes_ for AMS1117 regulator.    I know that _if_ it is present it will be AMS1117-3.3

Anyway,   it is all good news for the Display Shield.   It has correct components for _any_ Arduino.   e.g. 5V Uno or 3.3V Due.

You have some sort of problem with your Mega2560.   I don't know what.   After all,  the readreg output verifies that all the GPIO pins are working.

closed 2021-08-24 by prenticedavid

crow12gg, 2023-01-24:
This might be a little late but I too had this problem just now. The display worked fine on my UNO but only a bright white screen on the MEGA. I found the solution to be read the instructions. The PDF manual tells you to install the Elegoo display libraries from the disk to the libraries dir in your Arduino/libraries. I installed the Adafruit libraries for the display and the UNO worked fine. So the answer is to use the Elegoo display libs from the install libraries on the disk (or download from Elegoo) in the IDE libraries. Also, use the test sketches from the disk to do the testing since the Elegoo libs will be included in the code, not the Adafruit. Elegoo has different tests for the Uno than the Mega. Hope this helps somebody also having this problem.