prenticedavid/MCUFRIEND_kbv #177: 2.4" LCD white screen

halorocks982009, 2021-05-03:
//Read ID

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 06	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 02	Inversion Control
reg(0x00B6) 00 0A 82 27 04	Display Control
reg(0x00B7) 00 06	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 21 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 71	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 03 03	NVM Read
reg(0x00D3) 00 00 93 41	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 0F 16 14 0A 0D 06 43 75 33 06 0E 00 0C 09 08	GAMMA-P
reg(0x00E1) 00 08 2B 2D 04 10 04 3E 24 4E 04 0F 0E 35 38 0F	GAMMA-N
reg(0x00EF) 00 03 80 02 02 02	ILI9327
reg(0x00F2) 00 02 02 02 02 02 02 02 02 02 02 02	Adjust Control 2
reg(0x00F6) 00 01 00 00	Interface Control

Tried tft touchscreen and MCUFRIEND graphics test and enabling drivers in MCUFRIEND library and didn't work
Tried running tft.begin with different HEX, 0x9341 did not work either

prenticedavid, 2021-05-03:
1. Delete any manually installed library(s)
2. Install those libraries properly via IDE Library Manager.
3. Plug Shield into Uno.   All the male pins should engage with the corresponding female header sockets.
4. Run graphictest_kbv from examples
5. Copy-paste the Serial Terminal

Please answer each numbered step.  e.g. Y/N
Describe anything unusual or unexpected problem.

> `reg(0x00D3) 00 00 93 41 ILI9341, ILI9488`

Your readreg report looks 100% for an Ilitek ILI9341V controller.  ID should be reported as 0x9341

halorocks982009, 2021-05-03:
all libraries atm is installed via IDE library manager, the ' ' for include is for the previous students working on it
Shield is plugged properly 
Graphics test doesn't work
one arduino gave ID= 0x2121
Another gave ID=0x101

prenticedavid, 2021-05-03:
> the ' ' for include is for the previous students working on it
I do not understand.

I expect you to use an unchanged library.   And a regular Shield.    Please post a link to the actual shield that you have bought.
Or post a photo of front and back of the shield.
Or at least say what colour the pcb is.   And the part numbers printed on the visible integrated circuits.

David.

closed 2021-08-24 by prenticedavid

halorocks982009, 2021-08-24:
My Bad I did not know you relied but it’s okay now

Closed #177<https://github.com/prenticedavid/MCUFRIEND_kbv/issues/177>.