prenticedavid/MCUFRIEND_kbv #179: Where is println method used in MCUFRIEND_kbv/examples/aspect_kbv/aspect_kbv.ino

andrejmz, 2021-05-16:
Hi

It's more a question than an issue really, i apologise in advance if i lack c++ (that's not my main language) basic knowledge.
I cant find the definition of tft.println() in MCUFRIEND_kbv.cpp or Adafruit_GFX.cpp

best regards.

prenticedavid, 2021-05-17:
Adafruit_GFX inherits from Print (and Stream) classes.   e.g. overloaded print() and println() functions

Serial and many other "display" classes inherit from Print.h
Hence you get the same behaviour for `tft.print(1234, HEX)` as `Serial.print(1234, HEX)` as `lcd.print(1234, HEX)`

Multiple inheritance is one of the best things about C++
However it does mean that your target needs Print.h, WString.h, ... 

The actual mechanism that sends the print() characters to the output device is the write() function.

It also means that converting a C++ library to C gets messy.

David.

andrejmz, 2021-05-17:
Ok thanks!

closed 2021-05-17 by andrejmz