prenticedavid/MCUFRIEND_kbv #181: test_05A1 branch

prenticedavid, 2021-06-08:
1.  Switch to the "test_05A1" branch.
2.  run graphictest_kbv from examples
3.  it should force the ID to 0x05A1
4.  I hope that graphictest_kbv shows on the screen
5.  if not,  please copy-paste what it says on the Serial Terminal
6.  if you don't get a "white screen" take notes on paper.
7.  e.g. PORTRAIT is screen ribbon at the bottom,  note text direction 
8.  in Adafruits_report page,  note the ID value
9.  in Penguin screens,  note colours,  scroll directions
10. note "SOFTWARE SCROLL" or "readPixel error"  
11. note "INVERTED COLOURS" at the end of "GRAY Penguin" page
12. report back with the "problems". 
13. I expect that text is mirrored either horizontally or vertically

If you have cloned the library switching Branch is "easy"
If you have NOT cloned,  it means deleting the existing libraries/MCUFRIEND_kbv directory.
and "installing from ZIP"

Please ask if you are unsure of anything.   It is best to communicate via this "Issue"

I would really like to see the report from LCD_ID_readreg.ino

David.

zaffresky, 2021-06-08:
Sorry I opened a new issue instead of continuing this issue. graphictest_kbv results in a blank white screen. Here is the serial output:

```
Serial took 0ms to start
ID = 0x0
```

zaffresky, 2021-06-08:
Here is the serial output from  LCD_ID_readreg.ino:

```
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 71 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 77	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 F0 F0 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

```

prenticedavid, 2021-06-08:
I was expecting you to get some coloured graphics.  I presume that you are seeing my GitHub edits on your desk.

I have added an unlock-key to `examples/LCD_ID_readnew.ino`
Please run this.   See if you get something different from the readreg sketch.

Have you "cloned" or "forked" via GitHub ?
Or are you installing a library from ZIP ?

David.

zaffresky, 2021-06-08:
I am installing the library from zip. The output from the new sketch:

```
Read Special Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

diagnose any controller
reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 71 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 77	Get Pixel Format
reg(0x0030) 00 00 00 00 00	PTLAR
reg(0x0033) 00 00 00 00 00 00 00	VSCRLDEF
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 F0 F0 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B3) 00 00 00 00 00	Frame Memory
reg(0x00B4) 00 00	Frame Mode
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 80 80 10 00 00 00 00 00	Panel Control
reg(0x00C1) 00 1B 00 00	Display Timing
reg(0x00C5) 00 00	Frame Rate
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 0E 00 00	Power Control
reg(0x00D1) 00 00 00 00	VCOM Control
reg(0x00D2) 00 00 00	Power Normal
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 E3 10 1C 17 08 1D 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 3B 40 03 08 08 00 08 00 08 08 00	Adjust Control 2
reg(0x00F6) 00 04 00 08	Interface Control

```

prenticedavid, 2021-06-08:
The "unlock" has shown the contents of registers 0xC0 to 0xFF.
I was hoping to see an ID in 0xDA, 0xDB, 0xDC

Did you ever see anything with the Open Smart code ?

I will have a "think" tomorrow.
Your physical wiring is good.  Otherwise you would have never seen anything in LCD_ID_readreg or LCD_ID_readnew

line #97-98 in graphictest_kbv.ino should say : 
```
    ID = 0x05A1;                //force S6D05A1
    tft.begin(ID);
```
I would expect to see some colour or some flicker on the screen.

David.

zaffresky, 2021-06-08:
Yes, I did see some text with the Open-Smart code but it would be just for a second and then the screen would go blank. Every time I reuploaded the sketch or restarted the MCU I would see the same behavior. At that time I was using UNO with a logic level shifter. I first thought it may be due to the TXS0108E logic level converter. But even with a different level shifter nothing changed. 

Yeah, these lines are there in the graphics test sketch I have uploaded

```
    ID = 0x05A1;                //force S6D05A1
    tft.begin(ID);
```

zaffresky, 2021-06-09:
Good news. The graphics test is working properly 👍🏻 Reading from your instructions above to note the scroll directions, I have taken screenshots attached below.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

zaffresky, 2021-06-09:
I tried also the Touch_shield_new sketch but got just a black screen. Does that mean the touch screen is broken or unsupported driver as well?

prenticedavid, 2021-06-09:
I have added a couple of lines that should detect the ID.
Please install them.   Probably as easy to do it manually.

Now you can run all the example programs without any edits.
Please run the readpixels_kbv.ino example.   Take one photo of the screen.   e.g. white text on black background.

Thanks for confirming that _most_ of graphictest_kbv works.   
The only problem is **readPixel() ERROR.**

When you have posted the readpixels_kbv screenshot,  please edit your previous post.   Delete most of the photos.   Just leave ONE photo.

David.

prenticedavid, 2021-06-09:
I would like to get readPixel() correct first.

You run the Calibration sketch to test the Touch screen.   (if` readID()` is working correctly)
Everything should work fine on your 8MHz Pro Mini.

David.

zaffresky, 2021-06-09:
Here is the output from the readpixels sketch:

[image]

prenticedavid, 2021-06-09:
Please just edit MCUFRIEND_kbv.cpp i.e. change from
```

    case 0x05A1:
        _lcd_capable = AUTO_READINC | MIPI_DCS_REV1 | MV_AXIS | READ_24BITS;

```to
```

    case 0x05A1:
        _lcd_capable = AUTO_READINC | MIPI_DCS_REV1 | MV_AXIS;

```
I have not pushed to GitHub.  It is easier to wait for you to try readPixels_kbv.ino again.

David.

zaffresky, 2021-06-09:
Yep there is no readPixel() notification anymore :)  I have run the Calibration sketch. With a stylus the white crossbar doesn't turn red no matter where inside the crossbar I touch or even with a little more pressure. However, with my finger it does get red. Is it a hardware defect? The rest of the crossbars can easily be calibrated with a stylus

zaffresky, 2021-06-09:
Usually display driver and touch IC are separate but from my search so far it seems that S6D05A1 is a mobile phone display IC. Could that be that it also controls the touch function? I don't know it's possible to tell without the datasheet but I was wondering if that calibration issue could be related to using generic code instead of specific code for this IC. 

I could be totally wrong as I have very little experience with drivers and touch displays.

zaffresky, 2021-06-09:
Also when I run the readpixel sketch now it still has FFFF in green at several locations but different than before you fixed the readpixel() error.  Are these FFFFs in green normal or does it represent an error?

[image]

prenticedavid, 2021-06-09:
Brilliant.   The readPixel() / readGRAM() functions are working correctly in readpixels_kbv.ino
In graphictest_kbv.ino you should see SOFTWARE SCROLL moving horizontally across the screen.

There is no Touch controller chip.  Just the resistive panel.  The panel is read in software i.e. Touchscreen.h library

You should use a stylus for best results.   It is not unusual for the top right hand corner to be "difficult".  Just put your stylus as near to the cross-hairs as you can.
Copy-Paste the calibration lines from the Serial Terminal.
If you get an error or "unusual" warning,   run the sketch again.

David.

zaffresky, 2021-06-09:
With a stylus I can't make that crossbar turn red. It's only with a finger touch that it gets registered. Would it have an effect on the touch functionality in normal use?

Here's the output from the touch calib sketch:

```
TouchScreen.h GFX Calibration
Making all control and bus pins INPUT_PULLUP
Typical 30k Analog pullup with corresponding pin
would read low when digital is written LOW
e.g. reads ~25 for 300R X direction
e.g. reads ~30 for 500R Y direction

Testing : (A1, D7) = 33
Testing : (A2, D6) = 24
ID = 0x05a1

cx=884 cy=934 cz=344  X, Y, Pressure
cx=879 cy=527 cz=496  X, Y, Pressure
cx=876 cy=121 cz=595  X, Y, Pressure
cx=515 cy=926 cz=247  X, Y, Pressure
cx=511 cy=119 cz=505  X, Y, Pressure
cx=179 cy=920 cz=235  X, Y, Pressure
cx=143 cy=530 cz=352  X, Y, Pressure
cx=144 cy=119 cz=484  X, Y, Pressure

*** COPY-PASTE from Serial Terminal:
const int XP=6,XM=A2,YP=A1,YM=7; //320x480 ID=0x05A1
const int TS_LEFT=903,TS_RT=130,TS_TOP=943,TS_BOT=101;

PORTRAIT  CALIBRATION     320 x 480
x = map(p.x, LEFT=903, RT=130, 0, 320)
y = map(p.y, TOP=943, BOT=101, 0, 480)

LANDSCAPE CALIBRATION     480 x 320
x = map(p.y, LEFT=943, RT=101, 0, 480)
y = map(p.x, TOP=130, BOT=903, 0, 320)

```

prenticedavid, 2021-06-09:
Your Calibration report looks fine.

Copy the

```
const int XP=6,XM=A2,YP=A1,YM=7; //320x480 ID=0x05A1
const int TS_LEFT=903,TS_RT=130,TS_TOP=943,TS_BOT=101;
```

to TouchShield example.   Run the example.   Press the stylus in the **EXIT** box.   
Now you see a "Paint" program.  You can draw with the stylus.   It should correspond perfectly.
You might have difficulty with the top-right-hand-corner.

Please let me know about any problems.
An SD card should work too.   Connect it to 10, 11, 12, 13.

Have you asked Open-Smart for a datasheet ?

I will update the "current Beta" i.e. Master Branch on GitHub.   I just need to check if using this for ID is "safe"
`
reg(0x00A1) 00 F0 F0 00 00	RD_DDB SSD1963
`

David.

zaffresky, 2021-06-09:
The writing/drawing response on the screen is not good. Instead of a continuous line or a shape it just draws dots. With stylus or finger it doesn't matter. 

[image]

zaffresky, 2021-06-09:
Just checked my inbox on Aliexpress and they have sent the datasheet. Here it is:

[S6D05A1X01_DS_V0.00.pdf](https://github.com/prenticedavid/MCUFRIEND_kbv/files/6627120/S6D05A1X01_DS_V0.00.pdf)

prenticedavid, 2021-06-09:
Thanks for the datasheet.

I would expect a continuous line if you draw at writing speed.   Perhaps 8MHz makes a difference.
The Touch should work fine for buttons etc.

David.

zaffresky, 2021-06-09:
drawing at slower speed does increase number of dots but still it's not a continuous line. Guess it's the quality of resistive touch. 

I have read in the datasheet and also on the Aliexpress listing it mentions that the display has ram buffer which should take burden off the MCU and improve performance. It's around 518 kbytes of GRAM. Does the library make use of that?

Can I also use that GRAM in my sketches for improved performance?

prenticedavid, 2021-06-10:
Most LCD, OLED, TFT display controllers have internal GRAM (Graphics Ram)

They use this RAM when they update the pixels in a Scan line.   Much like a CRT TV draws each scan line from top to bottom and starts again.   In fact most TFT controllers interlace the scan lines just like a CRT TV.

So yes,  every time you drawPixel() or fillRect() you are updating the GRAM.

My library(s) have readPixel() and readGRAM() functions to read the GRAM.   readID() and readReg() to read controller registers.
Most Display Libraries only write to GRAM.   Many hardware display shields do not permit "read".

You can't really use this GRAM for non-display purposes.   You would "see" odd random pixels on the screen.   Or you disable some scan lines and see a "blank" area on the screen.

Thanks for the datasheet.   If you have personal communications with Open-Smart,  please tell them that you have contacted me.   I would be happy to support Open-Smart products.   e.g. I have your screen working with readID() and readPixel()
They can contact me via Github or by my private email.    
If you PM me via the arduino.cc Forum I will PM you with my private email that you can give it to your Open-Smart contact.

David.

zaffresky, 2021-06-10:
Hmm Ok. The way it was advertised by the Open-Smart was _"This display has a controller built into it with RAM buffer, so that almost all work is done by the TFT"_ so I thought this display is comparatively better than those with other driver IC's e.g. ILI9486 or ILI9481 etc. 

I will convey your message to Open-Smart and tell them how you helped me.  Personally Thank you very much for your time and support to get my display working 👍🏻 

In the beginning you also mentioned to use ESP32. Does your library support ESP32 or ESP8266?

zaffresky, 2021-06-10:
Also could you please tell how can I change the data pin (D0-D7) assignment to use other available pins?

closed 2021-08-24 by prenticedavid