prenticedavid/MCUFRIEND_kbv #183: ILI9325 on COVVY UNO R3 2.8" arduino touchscreen display

Mk1G, 2021-06-17:
First - thank you for the libraries. Amazing job, really appreciated.

I want to share a problem & my solution. I think the solution is a fudge, and will continue to try to understand the root cause.
I'd be interested to know if you've encountered something similar.

The display driver is an ILI 9325.
The specific board is the COVVY UNO R3 2.8 inch TFT Touch Screen, from amazon (uk)

1. I plugged the display into an Arduino Uno, using vanilla mcufriend_kbv libraries, and ran the display tests. Worked fine.
2. I plugged the display into a Nucleo-64 (F103RB) board, using the vanilla Controllers_Tech library based on your mcufriend_kvb libraries and ran the tests. Worked fine.
3. I plugged the display into my own board (STM32 L151 C8T6), with the same vanilla Controllers_Tech library and ran the tests. White screen.

Debugging my board showed that readID() was returning 0x9b2d. So data bit D3 was staying high. 
Using a logic analyser - I could see D3 moving up and down - so it wasn't shorted to another pin, or Vcc/Gnd.
If I stepped through the READ_8 method, (i.e. a very slow parallel read) it returned the correct 0x9325.

I plugged the display back into the Nucleo-64 board, and got the same problem. 
I.e. readID() returned 0x9b2d. Stepping through READ_8 returned the correct 0x9325.
I plugged the display back into the Arduino Uno - it worked fine. 

Back on the STM boards:
I implemented a delay(5000) in READ_8 between RD_STROBE; and read_8(); 
This solved the problem on both the Nucleo board and on my own board. The display tests ran perfectly.
Reducing the delay() to 1200 - problem returned. At 2500, the problem appeared intermittently.

I'll let you know if I learn more, and would appreciate any thoughts / suggestions for further investigation. Happy to provide more specific information if it would help. And thanks again for the amazing work on the libraries.

prenticedavid, 2021-06-17:
There is a Nucleo64-L152 board.   But no specific L151 support in the Arduino or even ST Eval or Discovery board.

I suggest that you add L152 to your mcufriend_special.h e.g to suit your wiring.
and you need to add another conditional block similar to
```
#elif defined(STM32L152xE)
#define WRITE_DELAY { } //32MHz M3
#define READ_DELAY  { RD_ACTIVE; }
#define GPIO_INIT()   { RCC->AHBENR |= RCC_AHBENR_GPIOAEN | RCC_AHBENR_GPIOBEN | RCC_AHBENR_GPIOCEN; }
#define PIN_OUTPUT(port, pin) PIN_MODE2((port)->MODER, pin, 0x1)
```

The easiest kludge is to use Nucleo64 wiring and hack mcufriend_shield.h
e.g. hack the IS_NUCLEO64 macro and hack the L152 condifional

I don't have an L152.  I have not checked the L151 datasheet.   I don't think that the L152 code has been tested.
Please provide feedback.

The L152 seems a bit crap compared to L476 (which I do have)

David.

Mk1G, 2021-06-17:
Hi David,
Thank you for coming back so quickly.
"The L152 seems a bit crap...". Yea, I know. JLCPCB was selling the L151 at $1.95 ea., and caught me in a moment of weakness...They are "not recommended for new projects" - although I haven't found any documented reason why they should be different from their mainstream cousins in my simple application.
Thanks for the L152 #define snippet. I was using the controllerstech dot com libraries, so I'll re-implement  with your mcufriend libraries directly, including that #define, and let you know how it goes.
Thanks again for a great set of libraries, and for your help & advice.Mk

prenticedavid, 2021-06-17:
Chip popularity amongst hobbyists is a bit of a mystery.  e.g. STM32F103
ST have much better, newr chips.  OTOH, BluePills are cheap from China.

L476 is a pretty versatile Low Power Chip.   You can run any chip SLOWLY.
But as a general rule,  you sleep most of the time and just wake up for swift bursts of regular processing speed.

David.

closed 2021-08-24 by prenticedavid