prenticedavid/MCUFRIEND_kbv #184: Touchscreen not working

TheSnowFall, 2021-07-04:
Hi, 
I am using 3.2 inches ili9341 240X320 TFT touch. With Arduino, it works very fine. The problem arises when I tried to integrate this with Stm Bluepill (64k). All graphics test works fine, but when I try uploading TouchScreen_Calibr_native through Arduino IDE to the blue pill there are some errors. I

Before starting with errors  I want to let you know that all the wiring was ok. I tested it.
IDE does not recognize A0 A5 A4 (132, 134 line no.) for the blue pill
I did not understand the for loop there, so I considered the A0 A4 A5 as Arduino pins and changed the PIN corresponding to Stm32 (compared to the position of those Arduino pins).  this is a work like Ameture I know Because I am one. Can anyone help me?

prenticedavid, 2021-07-04:
Post a link to the actual shield that you have bought.
Quote the wiring scheme.   e.g. recommended default BLUEPILL scheme or published "special" 

Have you installed via IDE Library Manager?
Or by Fork/Clone from GitHub ?

David.

TheSnowFall, 2021-07-04:
Thanks, Devid for your response. 

I Bought the uno shield version of the ILI9341 display from my local market.
I can share with you a similar product from banggood.
[https://www.banggood.com/2_8-Inch-TFT-LCD-Shield-Touch-Display-Screen-Module-Geekcreit-for-Arduino-products-that-work-with-official-Arduino-boards-p-989697.html?cur_warehouse=CN&rmmds=buy](url)

I am not using special. It is default . Here is my pin config:

> //LCD pins |D7 |D6 |D5 |D4 |D3 |D2 |D1 |D0 | |RD |WR |RS |CS |RST| |SD_SS|SD_DI|SD_DO|SD_SCK|
//STM32 pin |PA7|PA6|PA5|PA4|PA3|PA2|PA1|PA0| |PB0|PB6|PB7|PB8|PB9| |PA15 |PB5  |PB4  |PB3   | **ALT-SPI1**
 
I downloaded the library from GitHub as a zip(clone) and installed it to Arduino ide.

prenticedavid, 2021-07-04:
```
//LCD pins  |D7 |D6 |D5 |D4 |D3 |D2 |D1 |D0 | |RD |WR |RS |CS |RST| |SD_SS|SD_DI|SD_DO|SD_SCK|
//STM32 pin |PA7|PA6|PA5|PA4|PA3|PA2|PA1|PA0| |PB0|PB6|PB7|PB8|PB9| |PA15 |PB5 |PB4 |PB3 | ALT-SPI1
```
This looks like the default BLUEPILL wiring.

The Blue 2.8 inch shields tend to either have Touch Panel on LCD_D1, LCD_D0, LCD_RS, LCD_CS pins
or on LCD_D7, LCD_D6, LCD_WR, LCD_RS pins.

Just measure the resistance with a DMM.   Or run the Calibration sketch on a Uno.
It is not possible to determine the Touch pins in software.

David.

TheSnowFall, 2021-07-04:
So what you are saying is I have to find 4 pins and the p.x and p.y calibration value from Arduino. then I have to set corresponding pins for work out? Another question is how I will use Special blue pill pins?

prenticedavid, 2021-07-04:
Do you own a DMM ?
Do you know how to measure resistance ?

Since there are probably only two Touch schemes,   you could try both.
But it is not difficult to use a DMM.

David.

TheSnowFall, 2021-07-04:
> Do you own a DMM ?
> Do you know how to measure resistance ?
> 
> Since there are probably only two Touch schemes, you could try both.
> But it is not difficult to use a DMM.
> 
> David.

yes I have DMM. and also I know how to measure in ohms. thanks for your support

TheSnowFall, 2021-07-04:
> ```
> //LCD pins  |D7 |D6 |D5 |D4 |D3 |D2 |D1 |D0 | |RD |WR |RS |CS |RST| |SD_SS|SD_DI|SD_DO|SD_SCK|
> //STM32 pin |PA7|PA6|PA5|PA4|PA3|PA2|PA1|PA0| |PB0|PB6|PB7|PB8|PB9| |PA15 |PB5 |PB4 |PB3 | ALT-SPI1 [...]

 Hi David , I found the pins.
//LCD pins  | D1 | D0 |  RS |CS |
//STM32 pin |PA1|PA0 |PB7|PB8|

you mentioned in the examples that
`//BLUEPILL must have Analog for YP, XM`

here YP (CS> Lcd) and XM (RS>Lcd) are not in analog pin. please correct me if I am wrong.

prenticedavid, 2021-07-04:
**I would be much happier if you reported what the resistance value were.  i.e. ohms between which pins.**

Look at the Calibration sketch in the Beta.  (i.e. master Branch)
```
//int XP=PB7,XM=PA6,YP=PA7,YM=PB6; //BLUEPILL must have Analog for YP, XM
```
This has suitable defines for the common shields that use RS     D6     D7     WR

Note that you select PA0, PA1 for XM, YP because they are Analog pins
e.g. you select the appropriate statement.
```
//        RS     D6     D7     WR
//int XP=PB7,XM=PA6,YP=PA7,YM=PB6; //BLUEPILL must have Analog for YP, XM
//        CS     D1     D0     RS
//int XP=PB8,XM=PA1,YP=PA0,YM=PB7; //BLUEPILL must have Analog for YP, XM
```

I have just run the Calibration sketch with a regular red 2.4 shield and with a blue 3.5 shield

David.

TheSnowFall, 2021-07-04:
oh I forgot
D0>>CS was 0 ohm ...on pressed value increases
D1>>RS was 0 initially...on pressed value increases
additionally checked:
D0>>RS around 374 ohms...on pressed value increases
D1>>CS around 560 ohms...on pressed value increases

I tried again according to pin setup ... everything works except touch. I triple-checked my wiring. see result

https://user-images.githubusercontent.com/77330220/124397791-d0bc3d00-dd33-11eb-845a-25f34349cd41.mp4

And try out with Arduino mega result

https://user-images.githubusercontent.com/77330220/124397858-3f00ff80-dd34-11eb-80dd-44aaaaa45350.mp4

If I upload the "TouchScreen_Calibr_native" to arduino mega then this returns in serial:
`*** COPY-PASTE from Serial Terminal:
const int XP=8,XM=A2,YP=A3,YM=9; //240x320 ID=0x9341
const int TS_LEFT=924,TS_RT=113,TS_TOP=81,TS_BOT=907;

PORTRAIT  CALIBRATION     240 x 320
x = map(p.x, LEFT=924, RT=113, 0, 240)
y = map(p.y, TOP=81, BOT=907, 0, 320)

LANDSCAPE CALIBRATION     320 x 240
x = map(p.y, LEFT=81, RT=907, 0, 320)
y = map(p.x, TOP=113, BOT=924, 0, 240)`

touch is ok with Arduino mega/uno. But the problem occurs when I connect LCD to Stm32 with perfect wiring and pin. Only TFT works. touch does not.

one of the weird problems I ever encounter.

prenticedavid, 2021-07-05:
You have a Calibration result from your Mega2560
```
const int XP=8,XM=A2,YP=A3,YM=9; //240x320 ID=0x9341
const int TS_LEFT=924,TS_RT=113,TS_TOP=81,TS_BOT=907;
```
You have a wiring scheme for your BluePill
```
//LCD pins  |D7 |D6 |D5 |D4 |D3 |D2 |D1 |D0 | |RD |WR |RS |CS |RST| |SD_SS|SD_DI|SD_DO|SD_SCK|
//STM32 pin |PA7|PA6|PA5|PA4|PA3|PA2|PA1|PA0| |PB0|PB6|PB7|PB8|PB9| |PA15 |PB5 |PB4 |PB3 | ALT-SPI1
```

So all that you have to do is substitute the STM32 pin names e.g.
```
//const int XP=  8,XM= A2,YP= A3,YM=  9; //240x320 ID=0x9341 Mega pins
//const int XP=PA0,XM=PB7,YP=PB8,YM=PA1; //equivalent STM32 pins
//const int XM=PA0,XP=PB7,YM=PB8,YP=PA1; //ensure that XM, YP are STM Analog
const int XM=PA0,XP=PB7,YM=PB8,YP=PA1; //use the result
const int TS_LEFT=924,TS_RT=113,TS_TOP=81,TS_BOT=907;
```

I suggest that you run the TouchShield example with those two calibration lines.
Observe the ADC values reported for the stylus.   Click [EXIT].
Observe the stylus drawing in the "Paint" screen.

Life is much simpler with Uno/Mega because I can diagnose pins in software.
The Calibration sketch works on the STM32 if you tell it the XP, XM, ... pins
However the Calibration sketch does not report the XP, XM names correctly.   It does report the correct TS_LEFT, ... values correctly.

Incidentally,  everything works out of the box for a Nucleo board.   The Arduino A2 pin is correctly defined.   The Arduino D8 pin is correctly defined.
```
//const int XP=  8,XM= A2,YP= A3,YM=  9; //240x320 ID=0x9341 Mega pins
//const int XP= D8,XM= A2,YP= A3,YM= D9; //Nucleo An, Dn pins are defined for the Arduino headers
```

David.

TheSnowFall, 2021-07-05:
Hi, 
I tried those pin configs.  The screen becomes white(only the backlight is on) It should not happen. At least I should get any values.

prenticedavid, 2021-07-05:
I don't believe you.

If you have a problem,  please quote _which_ example sketch and copy-paste any "edits" e.g. calibration lines.

I make examples that should run "out of the box" on any board with proper Arduino headers e.g. Uno, Zero, Nucleo, ...
I used a DMM on this Blue 320x480 Shield:
```
LCD_D0, LCD_RS = 263R
LCD_D1, LCD_CS = 667R
```

This shield is running on a BluePill with the same wiring scheme as yours.

David.

TheSnowFall, 2021-07-05:
Ok, I will give another tryout and coming back to you again. Thanks for your continuous support.

TheSnowFall, 2021-07-06:
Hi @prenticedavid, 
sorry for disturbing again.
I want to run  "TouchScreen_Calibr_native " on Stm32. What is blocking me to compile this on Arduino ide is A0, A4, A5.(line 133, 135)
How can I get rid of this?

prenticedavid, 2021-07-06:
If you are using the "current Beta" i.e. master Branch,  the Calibration sketch avoids A0, A4, A5

If you are using the v2.9.9-Release you can just define some dummy values for A0, A4, A5

TheSnowFall, 2021-07-06:
Most Touch Screens use pins 6, 7, A1, A2
But they can be in ANY order
e.g. right to left or bottom to top
or wrong direction
Edit name and calibration statements

Please Calibrate.
ID=0x9341
Screen is 240x320
Calibration is: 
LEFT = 924 RT  = 113
TOP  = 81 BOT = 907
Wiring is always PORTRAIT
YP=24 XM=23
YM=1 XP=0

prenticedavid, 2021-07-06:
I suggest that you use Serial in your BluePill sketches.
You can configure USB CDC for Serial.
You can configure the HID bootloader.

```
YP=24 XM=23 means YP=PB8, XM=PB7
YM=1 XP=0 means YM=PA1, XP=PA0
```
That does not sound correct to me.  YP, XM **must** be Analog pins e.g. as I suggested earlier:

```
const int XM=PA0,XP=PB7,YM=PB8,YP=PA1; //use the result
```

If you find this too difficult,  I suggest that you buy a Nucleo board or Uno, Mega, ...
You would have the same task for Zero, Due, Teensy, ... i.e. 
```
const int XP= D8,XM= A2,YP= A3,YM= D9; //An, Dn pins are defined for the Arduino headers on Zero, Due, Teensy
```

David.

TheSnowFall, 2021-07-06:
David,

how do I use the special pin setup? should I call anything in the program?
Again I am thanking you for your support.

prenticedavid, 2021-07-06:
Just stick with the recommended wiring scheme.

Using a pre-existing "special wiring" just means connecting the wires correctly.   And enabling a USE_SPECIAL and a USE_MY_EXISTING_SPECIAL 

However writing a new "special" is a bit fiddly.

TheSnowFall, 2021-07-06:
> And enabling a USE_SPECIAL and a USE_MY_EXISTING_SPECIAL

Just wiring do all of it. I have no need to call in my program.
correct me if I am wrong. @prenticedavid

prenticedavid, 2021-07-06:
No,  you do not write anything different in your programs.
However,  **if** you had a write-only display,  you have to specify the correct ID because `readID()` does not work

TheSnowFall, 2021-07-06:
> Only enable the macros in the library files as "one-off".

elaborate, please

prenticedavid, 2021-07-06:
Just stick with the recommended wiring scheme.

TheSnowFall, 2021-07-06:
after wiring, it shows a white screen. pin scheme:

//LCD pins  |D7  |D6  |D5 |D4 |D3 |D2 |D1  |D0 | |RD |WR |RS |CS |RST| |SD_SS|SD_DI|SD_DO|SD_SCK| |SDA|SCL|
//STM32 pin |PA3 |PA2 |PA1|PA0|PB7|PB6|PA10|PA9| |PB1|PB0|PA7|PA6|PA5| |PB12 |PB15 |PB14 |PB13  | |PB9|PB8|

and in program:
`const int XP=PB9,XM=PA7,YP=PA6,YM=PB10; `

prenticedavid, 2021-07-07:
I used a DMM on this Blue 320x480 Shield to determine X, Y touch pins
```
LCD_D0, LCD_RS = 263R  e.g. X pins
LCD_D1, LCD_CS = 667R  e.g. Y pins
```

These are my Calibration reports for default BLUEPILL and for **special** MY_BLUEPILL
```
const int XP=PB7,XM=PA0,YP=PA1,YM=PB8; //320x480 ID=0x7796 BLUEPILL
const int TS_LEFT=124,TS_RT=907,TS_TOP=933,TS_BOT=57;

const int XP=PA9,XM=PA7,YP=PA6,YM=PA10; //320x480 ID=0x7796 MY_BLUEPILL
const int TS_LEFT=900,TS_RT=120,TS_TOP=62,TS_BOT=926;
```

Since your Shield seems to have the same X, Y pins as me,  I suggest that you wire as the **default BLUEPILL** and copy-paste my calibration lines into Touch_Shield_new.ino

Please report back with a TFT screen that displays graphics and a Touch panel that detects your stylus.

David.

closed 2021-08-24 by prenticedavid