prenticedavid/MCUFRIEND_kbv #185: ATMEGA328PB and Opensmart LCD

JackGlobetrotter, 2021-08-22:
I recently purchased this PCB with an atmega328pb and an TFT screen (possibly ili9325?). Having no experience coding for LCD registers i tried to set it up via this library, without success. The pinout seems to differ from the opensmart_uno shield and is as follows:
````````
#define LCD_RST 2 
[Serial TFT 2.4INCH with touch sreen Schematic.pdf](https://github.com/prenticedavid/MCUFRIEND_kbv/files/7027751/Serial.TFT.2.4INCH.with.touch.sreen.Schematic.pdf)
(PD2)
#define LCD_CS A3 (PC3, etc)
#define LCD_RS A2
#define LCD_WR A1
#define LCD_RD A0

#define LCD_D0 8
#define LCD_D1 9
#define LCD_D2 10 //10
#define LCD_D3 11 //11
#define LCD_D4 4
#define LCD_D5 13 //13
#define LCD_D6 6
#define LCD_D7 7

````````
the register output is as follows: 

````````
Read Special Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

diagnose any controller
reg(0x0000) 54 08	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) C4 C4 C4 C4	Manufacturer ID
reg(0x0009) C9 C9 C9 C9 C9	Status Register
reg(0x000A) CA CA	Get Power Mode
reg(0x000C) CC CC	Get Pixel Format
reg(0x0030) F0 F0 F0 F0 F0	PTLAR
reg(0x0033) F3 F3 F3 F3 F3 F3 F3	VSCRLDEF
reg(0x0061) E1 E1	RDID1 HX8347-G
reg(0x0062) E2 E2	RDID2 HX8347-G
reg(0x0063) E3 E3	RDID3 HX8347-G
reg(0x0064) E4 E4	RDID1 HX8347-A
reg(0x0065) E5 E5	RDID2 HX8347-A
reg(0x0066) E6 E6	RDID3 HX8347-A
reg(0x0067) E7 E7	RDID Himax HX8347-A
reg(0x0070) F0 F0	Panel Himax HX8347-A
reg(0x00A1) E1 E1 E1 E1 E1	RD_DDB SSD1963
reg(0x00B0) F0 F0	RGB Interface Signal Control
reg(0x00B3) F3 F3 F3 F3 F3	Frame Memory
reg(0x00B4) F4 F4	Frame Mode
reg(0x00B6) F6 F6 F6 F6 F6	Display Control
reg(0x00B7) F7 F7	Entry Mode Set
reg(0x00BF) FF FF FF FF FF FF	ILI9481, HX8357-B
reg(0x00C0) C0 C0 C0 C0 C0 C0 C0 C0 C0	Panel Control
reg(0x00C1) C1 C1 C1 C1	Display Timing
reg(0x00C5) C5 C5	Frame Rate
reg(0x00C8) C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8	GAMMA
reg(0x00CC) CC CC	Panel Control
reg(0x00D0) D0 D0 D0 D0	Power Control
reg(0x00D1) D1 D1 D1 D1	VCOM Control
reg(0x00D2) D2 D2 D2	Power Normal
reg(0x00D3) D3 D3 D3 D3	ILI9341, ILI9488
reg(0x00D4) D4 D4 D4 D4	Novatek
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) DC DC	RDID3
reg(0x00E0) E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0	GAMMA-P
reg(0x00E1) E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1	GAMMA-N
reg(0x00EF) EF EF EF EF EF EF	ILI9327
reg(0x00F2) F2 F2 F2 F2 F2 F2 F2 F2 F2 F2 F2 F2	Adjust Control 2
reg(0x00F6) F6 F6 F6 F6	Interface Control

````````

What i have tried so far to run the graphicstest_kbv :
1. Running without any changes -> white flashing screen
2. adding 328pb to mcufriend_special and mcufriend_shield and enabeling USE_OPENSMART_SHIELD_PINOUT_UNO    -> black screen
      - Adding SUPPORT_8347A  -> black screen
      - Adding SUPPORT_8352B -> black screen

I suspect that the different pinout is causing the troubles and i would be glad for any help. 

I have the schematics attached

prenticedavid, 2021-08-22:
`reg(0x0000) 54 08	ID: ILI9320, ILI9325, ILI9335, ...
`
You have a SPFD5408 controller.   The wiring is the same as Open-Smart shields.

Read paragraph #19 in C:\Users\ ... \Documents\Arduino\libraries\Mcufriend_kbv\extras\mcufriend_how_to.txt

So you need to enable:  ` USE_SPECIAL` and `USE_OPENSMART_SHIELD_PINOUT_UNO`

The SPFD5408 should be enabled by default.
Please let me know how you get on.

JackGlobetrotter, 2021-08-22:
hey, thanks for your help,
message is showing as intended, but LCD stays black this time...
Isn't the default opensmart pinout: control A0-A4.  Data D2-D9 ? Because mine uses also D10-13 for data....

prenticedavid, 2021-08-22:
Your wiring in the original post looks ok.  i.e. you received a valid ID.

The readreg report in your last post is pants !!

David.

JackGlobetrotter, 2021-08-23:
I agree, I will try again this evening after work.

JackGlobetrotter, 2021-08-23:
Hello again, continued testing this evening, first post readreg output is correct, the framework i used had a faulty pin assignment...
My problem does however persist...when i am using the correct read_8() and write_8 methods (mcufriend_special.h, the display stays black (the id is read correctly)
when using the default methods (mcufriend_shield.h), the display flickers white with different brightness modes....
I tried to change the RESET_PIN to the correct one without any results...
I don't know what to do anymore...

prenticedavid, 2021-08-23:
I have only added some comments to the special.h in the "master".
Genuine Open-Smart have a RESET chip and hence it is never connected to the Arduino headers,

Your schematic has a physical LCD_RST pin.   So I have amended for your board
```
#elif defined(__AVR_ATmega328P__) && defined(USE_OPENSMART_SHIELD_PINOUT_UNO)
//LCD pins  |D7 |D6 |D5 |D4 |D3 |D2 |D1 |D0 | |RD |WR |RS |CS |RST|
//AVR   pin |PD7|PD6|PB5|PD4|PB3|PB2|PB1|PB0| |PC0|PC1|PC2|PC3|PD2|
//UNO pins  |7  |6  |13 |4  |11 |10 |9  |8  | |A0 |A1 |A2 |A3 |2  |
#define RD_PORT PORTC
#define RD_PIN  0
#define WR_PORT PORTC
#define WR_PIN  1
#define CD_PORT PORTC
#define CD_PIN  2
#define CS_PORT PORTC
#define CS_PIN  3
#define RESET_PORT PORTD //n.c. on Open-Smart shields
#define RESET_PIN  2  // aliexpress.com/store/1199788

#define BMASK         B00101111
#define DMASK         B11010000
```

Do you have a genuine link for your display ?

JackGlobetrotter, 2021-08-23:
i got mine from here: 
https://www.dx.com/p/open-smart-2-4-uart-serial-tft-lcd-module-touch-screen-expansion-shield-with-touch-pen-for-arduino-uno-r3-mega2560-nano-2033627.html
the site has no firmware whatsoever (juste the arduino serial library...) so i wanted to play around with it to make a real "open" device....
already tried the changes for RST pin with no success..... i have also tried to disable UART as it may interfere with the LCD without success...

prenticedavid, 2021-08-23:
Thanks for the link.   I had never seen this before.    It seems to be an Open-Smart product.

It looks as if the ATmega328PB on the display pcb is an intelligent controller that receives UART commands from the Arduino.   It seems crazy to use an AVR.   Much easier and faster to use STM32 like a BluePill.

The 328PB firmware does not seem to be on the RAR file.   I suspect it will be locked.
If unlocked you could save the current firmware.  And replace with your own code.

I presume that you have run the original examples as shown in the video link.
And they perform the same as the video.

David.

JackGlobetrotter, 2021-08-24:
Hey, i do no tknow why the AVR was used, but i am trying to use the AVR to do more the just parse UART commands....it seemed to be a nice and easy board at first glance...
 I have saved the original firmware, but it does not give much away (as it is a binary file and searching for strings yields nothing of interest...). The original files work as intended with the base FW.
What i will do tomorrow is to verify the pinout given in the schematics, i suspect that something is off in there...Also the TFT may be an ILI9341, but i am unsure for now as the markings have been etched away:/
i will keep you updated.

prenticedavid, 2021-08-24:
Go on.   Most TFT controller chips are COG.  They are not visible to the human eye.

Your readreg report shows a SPFD5408 in 8080-8 interface mode.
I would expect MCUFRIEND_kbv to work.   

Your readreg #defines in the original post look correct.   The output is correct.   i.e. reg(0x0000) is ID.  most other registers are n.c. hence the "echo" response.

Please run readreg with your defines to verify your wires.
Then run graphictest_kbv.ino and observe ID reported on Serial.

Regarding backlight.  It requires an active-high signal on D3 (PD3)

Do you upload code into the 328PB via bootloader or external programmer ?
Have you cloned or forked this library?   Life is easier when I can just create a test Branch for you.
Do you have a USB-Serial dongle on TXD, RXD pins ?

David.

JackGlobetrotter, 2021-08-24:
Hello,
haven't done anything yet... Initially, I thought it would be easier, but if the project becomes more involved i will create a fork with pull request later on if that's fine with you? 
ID was reported correctly in readreg and graphicstrest_kbv.ino (which normaly hould mean the pinout is correct as the id is read via databus pins if i am not mistaken). 
I had enabled backlight via arduino pinMode and digitalWrite commands, but i will try again with c commands on avr level.

i am using usb-serial dongle (CH340 i belive, do not have it with me rn) connected to RX/TX (which are labeled RX_L TX_M in schematics). UART1 is diasbled (as it uses databus pins of tft). The board is programmed via UART or usbasp (i have tried both) with MicroCore bootloader (have tried pololu one too, but it is not as good).

JackGlobetrotter, 2021-08-28:
Hello,
here is the pr: https://github.com/prenticedavid/MCUFRIEND_kbv/pull/186
i added support for 328pb and implemented backlight if possible
Only thing i have not done is the following:
SPI (for SDcard) shares pins with LCD, so SPI has to be disabled to use TFT and vice-versa
``````````
void DisableSPI()
{
    noInterrupts();
    SPCR &= ~_BV(SPE);
    interrupts();
}

void EnableSPI()
{
    noInterrupts();
    SPCR |= _BV(SPE);
    interrupts();
}
``````````

I was not able to implement this in your lib, as it would change SPI to often (on every read and/or write)...
I may add an example in the BMP example ino

prenticedavid, 2021-08-28:
Regular Mcufriend shields don't have access to backlight.
Regular Open-Smart Shields have a PWM backlight that is disabled by default.

Your Open-Smart intelligent display is unusual.
My "RST pin" code should not interfere with regular Open-Smart Shields.

I am a little uneasy.  Perhaps the amended code should sit in a separate conditional block i.e. for 328PB.
After all,  the average user is not expected to access the 328PB chip. 

All Open-Smart Shields must disable / enable SPI on a Uno because of the shared pins.

I would prefer the User to handle SPI with beginTransaction() and endTransaction()
The User can control the PWM backlight as a user function rather than a library class method.

You have obviously Forked the Github library.   So your modifications will work ok for you.
If lots of readers purchase this 328PB Open-Smart board I might consider it.
If Open-Smart were to **request** smoother support for any of their Shields I would certainly help.

How are you getting on with your mods to the "intelligent UART display" ?

David.

closed 2022-02-12 by prenticedavid