prenticedavid/MCUFRIEND_kbv #187: test_IanBUK Branch

prenticedavid, 2021-08-29:
Please just change to MCUFRIEND_kbv/test_IanBUK Branch in your GitHub Desktop.

Run graphictest_kbv.ino from MCUFRIEND_kbv/test_IanBUK that Branch.   You need to close  graphictest_kbv.ino in the IDE.   Then reload from the examples.   (which will get you the MCUFRIEND_kbv/test_IanBUK branch

It looks as if my GitHub source files have got a few trailing spaces.   I will have to check where they have come from.    I sometimes edit in Notepad++, Keil, Rowley, AS7.0, ...
It makes comparisons very difficult.

But you clearly seem to have done a lot of "editing".   It is much easier if you don't make large changes until we have this working.
The advantage of using my MCUFRIEND_kbv/test_IanBUK branch is that you can't alter it.

If MCUFRIEND_kbv/test_IanBUK goes ok,  change to MCUFRIEND_kbv/test_IanBUK_8bit branch.
Compare the times in graphictest_kbv Adafruit_tests page.

David.

prenticedavid, 2021-08-29:
Please continue this GitHub subject on GitHub.

Since you have a GitHub account and a Fork,   the Arduino Forum is not appropriate.

> USE_MEGA_8BIT_SHIELD didn’t work - but USE_MEGA_8BIT_PORTC_SHIELD did.

USE_MEGA_8BIT_SHIELD uses db8..db15 on PORTA.   
USE_MEGA_8BIT_PORTC_SHIELD uses db0..db7 on PORTC.
USE_MEGA_16BIT_SHIELD uses  db0..db7 on PORTC, db8..db15 on PORTA.

Oops.  I had forgotten to select table for 7 inch SSD1963.   I have pushed these to both test branches.

Please just run graphictest_kbv.no and tell me what is wrong.
I don't want to second guess your programs.

I can fix all colours, directions, ... but only when I know what you are running and if you give me **accurate** feedback.
e.g. RED (portrait) Penguin screen shows "blue"
e.g. GREEN (landscape) Penguin screen has text that is horizontally mirrored.

I went to great trouble to make problems "obvious"  i.e. RED showing as a different colour like blue, yellow, cyan, ...

IanBUK, 2021-08-29:
David, thanks for your help.

the remaining problem - before I switch controllers to use the Nano 33 BLE - is that the RGB values are switched. So, effectively they are running as BGR rather than RGB.

The text appears to be the right way up, drawing is from the direction I expect, just the RGB values are wrong. What I draw as red appears blue, and vice-versa.

thanks,

Ian

prenticedavid, 2021-08-29:
I would really appreciate a proper answer.   i.e. which Penguin screens are wrong.
And whether the Penguin directions are correct.

When I get a reply I will push the **single** statement to both test branches.

David.

IanBUK, 2021-08-29:
I’ve finished on the laptop for the day now, I’ll try and get some time tomorrow.

IanBUK, 2021-08-29:
Ok, so here's a photo of the penguin, shown during the blue bands phase...

[image]

I've also videod the entire example test and will upload it to YouTube and share the link here, that will probably give a clearer view of what I'm getting. No chance for me miscommunicating it then. 

Ian

IanBUK, 2021-08-29:
YouTube video of the test sketch: https://youtu.be/pMQrglORA0I

prenticedavid, 2021-08-30:
1. Adafruit Tests: 15.75 secs
2. ID = 0xD3D3
3. BLUE (portrait) Penguin shows RED

1). I presume this is USE_MEGA_8BIT_PORTC_SHIELD.  I would expect USE_MEGA_16BIT_SHIELD to be faster.
2). write-only
3). it needs RGB instead of BGR.   I will remove the

IanBUK, 2021-08-30:
David,

The 15.7s timing was from 16bit mode, 8 bit mode (portC) took 36.25s. What sort of timings would you have expected?

I'll certainly give you feedback once I wire up the Nano 33 BLE. It's likely to be the weekend before I get chance to spend time on it now - it's going to be a busy week.

Your help is really appreciated.

Thanks.

Ian

prenticedavid, 2021-08-30:
Oh,  I suppose I could have calculated that for myself.   

> //#define USE_MEGA_8BIT_PORTC_SHIELD // 4.7sec Mega2560 Shield
> //#define USE_MEGA_16BIT_SHIELD     // 2.14sec Mega2560 Shield 
> 
If that was for a 240x320 on a MEGA2560 it implies 5 x 2.14sec = 10.7sec for 800x480 (16-bit) and 23.5sec for 8-bit.
Perhaps I should dig out a USE_MEGA_16BIT_SHIELD and a USE_MEGA_8BIT_PORTC_SHIELD and check those (historic timings)

Don't worry about the times for the moment.  Just get the colours and Read functions working (on BLE33).

Time becomes important on bigger screens.   You need to minimise any update areas.

Life is MUCH easier with branches on a Fork !!
The average Arduino user just installs a Release.   This means you have to "trust" that they followed any edits you may suggest.

David.

IanBUK, 2021-09-05:
I've spent this afternoon wiring up the nano 33 ble to use with the shield. Triple checked my wiring, used a multi-meter to check that I'm not getting shorts. 

But...the backlight doesn't come on. I can't guarantee what the problem is. My thoughts are around the following possibilities:

1) The backlight is simply not coming on
2) The screen isn't getting the power it needs
3) My wiring is stuffed or I've got a short
4) As well as wiring up the power to the +5v on the main slot, I also need to wire up ground an 5v on the other connectors.

The code is running - I added some output to loop() and it was getting churned out to Serial.

It's time to stop for the day now. Next weekend, I'll have to look at my wiring again. Apart from that, I could try putting a scope on the signals to see what's going across to the board.

Sorry it's not more positive news - but I'm sure that the problem is my end. 

Ian

prenticedavid, 2021-09-05:
You just run LCD_ID_readreg.ino

LCD_ID_readreg.ino does not care about backlight.   You just check that there is some sensible readings.   This will verify your wiring.

The backlight should be controlled by a transistor.  You just apply a logic high or a PWM signal.   Nothing to do with the TFT library.

David.

IanBUK, 2021-09-05:
This is the output from LCD_ID_readreg.ino:

```
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) FF FF	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 04 04 04 04	Manufacturer ID
reg(0x0009) 09 01 01 01 01	Status Register
reg(0x000A) 0A 02	Get Power Mode
reg(0x000C) 0C 04	Get Pixel Format
reg(0x0061) 61 61	RDID1 HX8347-G
reg(0x0062) 62 62	RDID2 HX8347-G
reg(0x0063) 63 63	RDID3 HX8347-G
reg(0x0064) 64 64	RDID1 HX8347-A
reg(0x0065) 65 65	RDID2 HX8347-A
reg(0x0066) 66 66	RDID3 HX8347-A
reg(0x0067) 67 67	RDID Himax HX8347-A
reg(0x0070) 70 70	Panel Himax HX8347-A
reg(0x00A1) A1 A1 A1 A1 A1	RD_DDB SSD1963
reg(0x00B0) B0 B0	RGB Interface Signal Control
reg(0x00B4) B4 B4	Inversion Control
reg(0x00B6) B6 B6 B6 B6 B6	Display Control
reg(0x00B7) B7 B7	Entry Mode Set
reg(0x00BF) BF BF B7 B7 B7 B7	ILI9481, HX8357-B
reg(0x00C0) C0 C0 C0 C0 C0 C0 C0 C0 C0	Panel Control
reg(0x00C8) C8 C8 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0	GAMMA
reg(0x00CC) CC CC	Panel Control
reg(0x00D0) D0 D0 D0	Power Control
reg(0x00D2) D2 D2 D2 D2 D2	NVM Read
reg(0x00D3) D3 D3 D3 D3	ILI9341, ILI9488
reg(0x00D4) D4 D4 D4 D4	Novatek ID
reg(0x00DA) DA DA	RDID1
reg(0x00DB) DB DB	RDID2
reg(0x00DC) DC DC	RDID3
reg(0x00E0) E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0	GAMMA-P
reg(0x00E1) E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1	GAMMA-N
reg(0x00EF) EF EF E7 E7 E7 E7	ILI9327
reg(0x00F2) F2 F2 F2 F2 F2 F2 F2 F2 F2 F2 F2 F2	Adjust Control 2
reg(0x00F6) F6 F6 F6 F6	Interface Control
```

IanBUK, 2021-09-05:
I might have a clue. Last week I had to add the following lines, to turn on the backlight:

```
pinMode(8, OUTPUT);  //backlight
digitalWrite(8, HIGH);//on
```

That worked great on the Mega. Now that I've moved over to the Nano, this doesn't work. I suspect this is because pin 8 is now doing something different and pin 8 on the oshield is connected to ... nothing. I suspect that if I wire it up to a pin on the Nano 33 BLE and set it to OUTPUT and HIGH, it might even work.

Not tonight though. I've packed the soldering iron and stuff away, and I think I've wrecked the sobering iron tip....Yeh, I'm a beginner there too.

Ian

prenticedavid, 2021-09-06:
`reg(0x00A1) A1 A1 A1 A1 A1	RD_DDB SSD1963
`

Your wiring is wrong.   I would expect to see an ID value here.  e.g.
`reg(0x00A1) 01 57 61 01 FF  RD_DDB SSD1963
`

Regarding the backlight.   Modules often have a weak pullup which means the backlight will be "on" with n.c.
In your case it appears that you need an external logic high.

David.

IanBUK, 2021-09-12:
Well that went well*

*no, it really didn’t.

I got to the end of yesterday and whilst lcd_id_readreg still reported the same values, I was able to get the screen to turn on.

today, I went through and wired up the miso/mosi/sck etc. Thinking that I must have missed them and that I was close to having it working.

no.

With my ‘shield’ and the Arduino, lcd shield and lcd plugged in, the Arduino doesn’t power up. If I take the Arduino out of the board, it powers. If I put the Arduino back in the board and unplug the shield and lcd, it powers. If I put the shield and screen back in the Mega, lcd_id_readreg returns the same values as last week, but nothing is displayed when I run the graphics test - the backlight flickered a couple of times then stopped. 

I’ve given up for the day. At best, I need to take out the wires since the last successful test. At worst, I’ve blown the expensive screen.

ian

prenticedavid, 2021-09-12:
Change to the "master" Branch and wire your NANO BLE33 according to the mcufriend_shield.h

```
//################################## NANO33 BLE ############################
#elif defined(ARDUINO_ARDUINO_NANO33BLE)
#warning regular UNO shield on a Nano33 BLE

//LCD pins  |D7   |D6   |D5   |D4   |D3   |D2   |D1   |D0   | |RD  |WR  |RS   |CS   |RST  |
//BLE pin   |P0.23|P1.14|P1.13|P1.15|P1.12|P1.11|P0.27|P0.21| |P0.4|P0.5|P0.30|P0.29|P0.31|
//NANO pins |7    |6    |5    |4    |3    |2    |9    |8    | |A0  |A1  |A2   |A3   |A4   |
```

That looks a straightforward scheme.  Obviously you connect 3.3V and GND too.

You could add another comment line for each wire colour.   Line it up nicely to fit the other lines.
Post a photo of your actual wires.

This sounds like about 21 minutes of your life.   e.g. 10 mins to document the wire colours.  10 minutes to attach the coloured Dupont cables.   1 minute to post a photo.

I have no intention of making a Protoshield if you are not prepared to spend that "20 minutes".   I don't want to waste a 20x2 header socket.

IanBUK, 2021-09-12:
David,

I will certainly do that. It might well be the weekend before I get to spend any time on it. 

Ian

IanBUK, 2021-09-13:
Well, sleep evaded me once again. Thanks neuro-pain, I really appreciate it. No, really, I mean, who needs to sleep before a week at work?

/whingeOver

So, yes, some of my wiring was wrong. WR & CS were swapped. However, on reading the data sheet again, to get the 'A' pins:

```
//LCD pins  ||RD  |WR  |RS   |CS   |RST  |
//BLE pins ||P0.4|P0.5|P0.30|P0.29|P0.31|
//NANO pins | |A0  |A1  |A2   |A3   |A4   |

```
I was checking against the pins in the data sheet:

```
|No|Symbol| Descriptions|
|-- | -- | --|
|1| +5V|+5V Power|
|2|+5V|+5V Power|
|3|DB8(IO22)|Data bus 8 to Arduino Board Digital IO 22|
|4|DB9(IO23)|Data bus 9 to Arduino Board Digital IO 23| 
|5|DB10(IO24)|Data bus 10 to Arduino Board Digital IO 24|
|6|DB11(IO25)|Data bus 11 to Arduino Board Digital IO 25|
|7|DB12(IO26)|Data bus 12 to Arduino Board Digital IO 26|
|8|DB13(IO27)|Data bus 13 to Arduino Board Digital IO 27|
|9| DB14(IO28)|Data bus 14 to Arduino Board Digital IO 28|
|10|DB15(IO29)|Data bus 15 to Arduino Board Digital IO 29|
|11|DB7(IO30)|Data bus 7 to Arduino Board Digital IO 30|
|12|DB6(IO31)|Data bus 6 to Arduino Board Digital IO 31|
|13|DB5(IO32)|Data bus 5 to Arduino Board Digital IO 32|
|14|DB4(IO33)|Data bus 4 to Arduino Board Digital IO 33|
|15|DB3(IO34)|Data bus 3 to Arduino Board Digital IO 34|
|16|DB2(IO35)|Data bus 2 to Arduino Board Digital IO 35|
|17|DB1(IO36)|Data bus 1 to Arduino Board Digital IO 36|
|18|DB0(IO37)|Data bus 0 to Arduino Board Digital IO 37|
|19|D/C(IO38)|Data/Command Select,Arduino Board Digital IO 38|
|20|/WR(IO39)|Write Strobe Signal , Arduino Board Digital IO 39|
|21|LCD_/CS(IO40)|Chip Select Signal, Arduino Board Digital IO 40|
|22|/RESET|LCD Reset Signal, Arduino Board Digital IO 41|
|23|RTP_CS|Resistive Touchscreen Chip Select Signal, Arduino Board Digital IO 42|
|24|SCLK Serial Clock signal, Arduino Board Digital IO 43 (SD Card, RTP,Falsh,Font Chip)|(Software SPI)|
|25|DIN (SD card, RTP,Falsh,font chip)|Serial Data Input, Arduino Board Digital IO 44 (Software SPI)|
|26|Dout (SD card, RTP,Falsh,font chip)|Serial Data Output, Arduino Board Digital IO 45 (Software SPI)|
|27|RTP PEN|Resistive Touch Screen Pen Interrupt, Arduino Board Digital IO 46|
28|SD_CS|SD Card Chip Select Signal, Arduino Board Digital IO 47|
|29 | CTP INT | Capacitive Touch Screen Interrupt, Arduino Board Digital IO 48|
|30 |NC|No Connect |  
|31 |NC |No Connect
|32 |NC |No Connect
|33 |NC |No Connect |  
|34|NC |No Connect
|35 |VSS |Ground|
|36 |VSS |Ground |  
```
So:
A1 (WR) -> pin 20
A3 (CS) -> pin 21
A4 (RST) -> pin 22

That leaves A0 (RD) and A2(RS) which I'm not sure what to connect to as I really can't see in that list which pins they should connect to on the shield.

Got to go now - my work alarm has gone off.

Ian

prenticedavid, 2021-09-13:
Go on.  The Nano boards all have the Analog pins and Digital pins clearly marked.  So if I say connect RD pin to A0 you simply attach a Dupont from RD on the SSD1963 board to A0 on the Nano board.  And if you are using an Orange cable for RD and a Purple cable for DB7 say:

```
//LCD pins  |D7   |D6   |D5   |D4   |D3   |D2   |D1   |D0   | |RD  |WR  |RS   |CS   |RST  |
//BLE pin   |P0.23|P1.14|P1.13|P1.15|P1.12|P1.11|P0.27|P0.21| |P0.4|P0.5|P0.30|P0.29|P0.31|
//NANO pins |7    |6    |5    |4    |3    |2    |9    |8    | |A0  |A1  |A2   |A3   |A4   |
//wire color|Purpl|Blue |5    |4    |3    |2    |9    |8    | |Orng|A1  |A2   |A3   |A4   |
```
Yes,  it makes life easier when there is a neatly formatted table.   It helps you and it helps your readers.
Personally I find printed names like A1 or WR to be convenient when attaching Dupont cables.   DC, D/C, RS all mean the same thing.
Please post a link to the actual Screen that you have bought.   It is probably on the Arduino Forum but it would be handy to have it on the GitHub Issues.  I am sure that they all have RD, WR, DC, CS, RESET pins although DC might have different "name"

[image]

IanBUK, 2021-09-13:
David,

The actual screen is: https://www.buydisplay.com/7-inch-arduino-touch-screen-shield-ssd1963-library-for-mega-due

Thanks,

Ian

prenticedavid, 2021-09-13:
I assume that your display has a 20x2 male header and not the ZIF ribbon connector.

You will be connecting the BLE33 to the 20x2 header.   You don't need the Adapter Shield for the BLE33.

Since you already have the BuyDisplay Adapter Shield you might just as well plug it into a Mega2560 or a Due.   Why bother with a BLE33 ?

Note that BuyDisplay 40-pin is different to the regular Ebay/Aliexpress 40-pin Display.

David.

IanBUK, 2021-09-15:
Morning David,

Yes, the display has the 20x2 header rather than the ZIF ribbon connector.

Why the Nano 33 BLE rather than the Mega2560? A few reasons: processor speed, memory, storage and my project’s need to communicate with multiple BLE devices. 

I’ll add more updates, including photos as soon as I can. I’ve just gone back to the basic of setting up the screen, shield and Mega2560 to prove I hadn’t destroyed the screen - it’s still working.

Ian

IanBUK, 2021-09-15:
Sorry about the table layout by the way, it looked ok on my screen when I sent the message.

IanBUK, 2021-09-18:
OK, so wiring directly to the screen rather than via the shield, I get:

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 08 08	Get Power Mode
reg(0x000C) 00 00	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 01 57 61 01 FF	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 01 EF	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

IanBUK, 2021-09-18:
[image]

Mandelbrot set, rendered on Nano 33 BLE, displayed on 7" ssd1963 800x480 display in 8bit mode.

IanBUK, 2021-09-18:
Now to work out the touch and SD stuff....

prenticedavid, 2021-09-19:
I am pleased that you have got it working.   I hope that my advice in https://github.com/prenticedavid/MCUFRIEND_kbv/issues/187#issuecomment-917891633 was helpful.

I would appreciate some feedback.   e.g. numbers from the Adafruit Tests Report page.

If you want to use the SD and the XPT2046 you should connect to the BLE33 hardware SPI pins.
And experience just how crap the BLE33 SPI class implementation is.

I was considering soldering up a specific SSD1963 Protoshield.   So that I could test SSD1963 in 8-bit mode on AVR, Teensy, BLE33, ESP32, ...
Now that you are successful,  I don't need to.

David.

IanBUK, 2021-09-19:
David, 

yes - your advice was very helpful thank you. I’ll post the Adafruit numbers later today. I may even get as far as wiring up the XPT2046 - the SD card is less important for my device. 

I think I owe you a beer or something!

Thanks,

ian

IanBUK, 2021-09-19:
@prenticedavid stats below from running graphictest_kbv on a Nano 33 BLE with the 7" SSD1963 from BuyDisplay:

[image]

prenticedavid, 2021-09-19:
Looking at the shield.h you could adjust the delays for a SSD1963.   Which would make a dramatic improvement to the Write (and Read) times.

```
//################################## NANO33 BLE ############################
#elif defined(ARDUINO_ARDUINO_NANO33BLE)
#warning regular UNO shield on a Nano33 BLE

//LCD pins  |D7   |D6   |D5   |D4   |D3   |D2   |D1   |D0   | |RD  |WR  |RS   |CS   |RST  |
//BLE pin   |P0.23|P1.14|P1.13|P1.15|P1.12|P1.11|P0.27|P0.21| |P0.4|P0.5|P0.30|P0.29|P0.31|
//NANO pins |7    |6    |5    |4    |3    |2    |9    |8    | |A0  |A1  |A2   |A3   |A4   |

#define WRITE_DELAY { WR_ACTIVE8; }   //M4F @ 64MHz
#define IDLE_DELAY  { WR_IDLE2; }
#define READ_DELAY  { RD_ACTIVE4; RD_ACTIVE; }

```

IanBUK, 2021-09-19:
Getting the text from that:

```
MCUFRIEND 3.00 UNO
FillScreen                1758018
Text                      47544
Lines                     2319223
Horiz/Vert Lines          145482
Rectangles (outline)      76353
Rectangles (filled)       5735911
Circles (filled)          700559
Circles (outline)         958280
Triangles (outline)       395227
Triangles (filled)        1963932
Rounded rects (outline)   283654
Rounded rects (filled)    5754700
Total: 20.14sec
ID: 0x1963
MHz -0s
```

IanBUK, 2021-09-19:
> Looking at the shield.h you could adjust the delays for a SSD1963. Which would make a dramatic improvement to the Write (and Read) times.
>
> ```
> //################################## NANO33 BLE [...]

Changing the code to: 
```
#define WRITE_DELAY { WR_ACTIVE2; }   //M4F @ 64MHz
#define IDLE_DELAY  { WR_IDLE2; }
#define READ_DELAY  { RD_ACTIVE2; RD_ACTIVE; }
```

resulted in: 

```MCUFRIEND 3.00 UNO
FillScreen                1191631
Text                      41775
Lines                     2181950
Horiz/Vert Lines          99562
Rectangles (outline)      53086
Rectangles (filled)       38891347
Circles (filled)          553915
Circles (outline)         887228
Triangles (outline)       368356
Triangles (filled)        1383657
Rounded rects (outline)   248035
Rounded rects (filled)    3921152
Total: 14.82ec
ID: 0x1963
MHz -0s
```

IanBUK, 2021-09-19:
Taking the WRITE_DELAY any lower results in on-screen corruption and instability.  Putting it back to WR_ACTIVE_2 and reducing the IDEL_DELAY to WR_IDLE resulted in a total of 16.2s. So, it looks like the best combination is:

```
#define WRITE_DELAY { WR_ACTIVE2; }   //M4F @ 64MHz
#define IDLE_DELAY  { WR_IDLE2; }
#define READ_DELAY  { RD_ACTIVE2; RD_ACTIVE; }
```

prenticedavid, 2021-09-19:
So 2 : 1 : 3 is a healthy delay combination for your SSD1963.
Note that 8080-16 interface has a significant optimisation for fillRect().
But 16.2s is not unreasonable for an "Arduino"

David.

closed 2022-02-12 by prenticedavid