prenticedavid/MCUFRIEND_kbv #188: force ID 0x9341 on 0xD3D3

rvxfahim, 2021-09-05:
How can I force 0x9341 with the guislice library and mcufriend library selected as the driver?
I can see that in examples of MCUFRIEND diagnose touch, there is a function where readID returns D3D3 and chooses ID = 0x9486 but I replace this value with 0x9341 and then the touch calibration works fine. It is a 2.4" display which is bought from local electronics store and hence I can't provide link. The packet only had a written instruction as "0x9341 compatible".
If I can just force 0x9341 through GUIslice library then my existing project will work fine because I already made a bunch of prototypes with 0x9341 displays. Now after burning the program the display remains white with the new batch of 2.4" LCDs

prenticedavid, 2021-09-05:
All "Mcufriend Uno Shields" are read-write.   So I do not expect you to see 0xD3D3.

If it is a "Uno Shield" run  LCD_ID_readreg.ino from the examples.   Copy-paste to your message.

The current "Beta" supports some extra IDs that were not known at the time of v2.9.9-Release (as installed by the IDE Library Manager)

David.

rvxfahim, 2021-09-11:
I am going to collect the readreg info and post here asap.

rvxfahim, 2021-09-12:
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)
reg(0x0000) C0 C0	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 08 08	Get Power Mode
reg(0x000C) 06 06	Get Pixel Format
reg(0x0061) E1 E1	RDID1 HX8347-G
reg(0x0062) E2 E2	RDID2 HX8347-G
reg(0x0063) E3 E3	RDID3 HX8347-G
reg(0x0064) E4 E4	RDID1 HX8347-A
reg(0x0065) E5 E5	RDID2 HX8347-A
reg(0x0066) E6 E6	RDID3 HX8347-A
reg(0x0067) E7 E7	RDID Himax HX8347-A
reg(0x0070) F0 F0	Panel Himax HX8347-A
reg(0x00A1) E1 E1 E1 E1 E1	RD_DDB SSD1963
reg(0x00B0) F0 F0	RGB Interface Signal Control
reg(0x00B4) F4 F4	Inversion Control
reg(0x00B6) F6 F6 F6 F6 F6	Display Control
reg(0x00B7) F7 F7	Entry Mode Set
reg(0x00BF) FF FF FF FF FF FF	ILI9481, HX8357-B
reg(0x00C0) C0 C0 C0 C0 C0 C0 C0 C0 C0	Panel Control
reg(0x00C8) C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8	GAMMA
reg(0x00CC) CC CC	Panel Control
reg(0x00D0) D0 D0 D0	Power Control
reg(0x00D2) D2 D2 D2 D2 D2	NVM Read
reg(0x00D3) D3 D3 D3 D3	ILI9341, ILI9488
reg(0x00D4) D4 D4 D4 D4	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0	GAMMA-P
reg(0x00E1) E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1	GAMMA-N
reg(0x00EF) EF EF EF EF EF EF	ILI9327
reg(0x00F2) F2 F2 F2 F2 F2 F2 F2 F2 F2 F2 F2 F2	Adjust Control 2
reg(0x00F6) F6 F6 F6 F6	Interface Control

@prenticedavid this is LCD_ID_readreg.ino

rvxfahim, 2021-09-12:
Diagnose whether this controller is supported
There are FAQs in extras/mcufriend_how_to.txt
tft.readID() finds: ID = 0xD3D3
MCUFRIEND_kbv version: 2.9.9
Probably a write-only Mega2560 Shield
Try to force ID = 0x9481
PORTRAIT is 320 x 480
Run the examples/graphictest_kbv sketch
All colours, text, directions, rotations, scrolls
should work.  If there is a problem,  make notes on paper
Post accurate description of problem to Forum
Or post a link to a video (or photos)
I rely on good information from remote users

This is from diagnose TFT support
@prenticedavid

prenticedavid, 2021-09-12:
Thanks for the readreg report.

```
reg(0x0004) 00 00 00 00 Manufacturer ID
...
reg(0x00D3) D3 D3 D3 D3 ILI9341, ILI9488
...
reg(0x00DA) 00 00 RDID1
reg(0x00DB) 00 00 RDID2
reg(0x00DC) 00 00 RDID3
...

```
So there does not seem to be any _likely_ ID at all.   But it is a "MIPI-style" controller

I suggest that you run graphictest_kbv.ino
just force ID = 0x9329 in setup()
report all the "errors" e.g.
wrong colours,
mirror text.
wrong scroll directions.
readPixel() error.

I will tell you the "best" ID to try.

David.

rvxfahim, 2021-09-12:
0x9329 - worked with no errors
0x9341- worked with no errors

prenticedavid, 2021-09-12:
I suggest that you always use this sequence in your setup()

```
    uint16_t ID = tft.readID();
    if (ID == 0x0000) ID = 0x9329;  //the current Beta should return 0x0000
    if (ID == 0xD3D3) ID = 0x9329;  //the v2.9.9 Release returns 0xD3D3
    tft.begin(ID);
```

I suspect that there is a secret unlock key e.g. in reg(0xB0)
You can try different "unlock keys" in the LCD_ID_readnew.ino

David.

rvxfahim, 2021-09-12:
> I suggest that you always use this sequence in your setup()
> 
> ```
>     uint16_t ID = tft.readID();
>     if (ID == 0x0000) ID = 0x9329;  //the current Beta should return 0x0000
>     if (ID == 0xD3D3) ID = 0x9329;  //the v2.9.9 Release returns 0xD3D3
>     tft.begin(ID);
> ```
> 
> I suspect that there is a secret unlock key e.g. in reg(0xB0)
> You can try different "unlock keys" in the LCD_ID_readnew.ino
> 
> Are you sure that 0x9341 "works" 100% ? e.g. "SOFTWARE SCROLL"
> Please can you tell me what ID is reported from the current Beta (i.e. master Branch)
> 
> David.

by current Beta master branch do you mean 2.9.9 in the release section? If yes then it returns ID as 0xD3D3
I checked again there were no errors by forcing both 9341 and 9329. Software scroll worked on both.

prenticedavid, 2021-09-12:
Please confirm that readID() returns 0x0000 in the Beta.

Re LCD_ID_readnew.ino
You can enable different "unlock keys" which _might_ reveal extra register values.

If I can differentiate between your 0x0000 and another controller I can support your controller transparently.