prenticedavid/MCUFRIEND_kbv #189: blue pill (stm32f104 )- 8 bit Parallel - 2.4 inch tft - ILI9341

madhavanatmel, 2021-09-25:
hi, i trying to connect the blue pill (stm32f104 )-

2.4 inch tft - ILI9341-8 bit Parallel mode but no output in display .
i am trying use display module as Only Display NO touch requirement at this movement 
the following steps i have followed :
1. wired the display in below pattern 
2. //LCD pins  |D7 |D6 |D5 |D4 |D3 |D2 |D1 |D0 |           |RD |WR |RS |CS |RST|
  //STM32 pin |PA7|PA6|PA5|PA4|PA3|PA2|PA1|PA0|    |PB0|PB6|PB7|PB8|PB9| 

3. I have uploaded Glue_Demo_320x240 sketechs without any modification.

 display giving white screen and after some time some  black and white dots (like Old CRT  television noise without video signal) is coming  . white screen and noisy screen is changing is 3 sec interval .

Suggest me to how to get connect . 

additionally one more request it's possible to change the PIN number because i need the use SPI memory device in SPI 1 port default PA7 - PA4 

Using the official STM32 Duino COre

prenticedavid, 2021-09-25:
I have no idea what a stm32f104 is.

However from mcufriend_shield.h:

```
#elif defined(ARDUINO_GENERIC_STM32F103C) || defined(ARDUINO_BLUEPILL_F103C8) || defined(ARDUINO_BLUEPILL_F103CB)
#warning Uno Shield on BLUEPILL

//LCD pins  |D7 |D6 |D5 |D4 |D3 |D2 |D1 |D0 | |RD |WR |RS |CS |RST| |SD_SS|SD_DI|SD_DO|SD_SCK|
//STM32 pin |PA7|PA6|PA5|PA4|PA3|PA2|PA1|PA0| |PB0|PB6|PB7|PB8|PB9| |PA15 |PB5  |PB4  |PB3   | **ALT-SPI1**

```

What settings have you made in the Arduino IDE ?

David.

madhavanatmel, 2021-09-25:
oh oh i am sorry it's typing mistake it's stm32f103 .

 Arduino Setting Mean ?
[image]

prenticedavid, 2021-09-25:
Copy-paste the settings in your Tools menu e.g.
Board:
Board part number:
USART:
USB:

madhavanatmel, 2021-09-25:
[image]

prenticedavid, 2021-09-25:
I would expect USB CDC and HID Bootloader.   Or perhaps Upload = SWD.
But your settings should work just fine with Upload = Serial (if you have a USB-Serial adapter)

I can only suggest that you check your wires.
A "good strategy" is to add your wire colors to the scheme e.g.

```
//LCD pins  |D7 |D6 |D5 |D4 |D3 |D2 |D1 |D0 | |RD |WR |RS |CS |RST| |SD_SS|SD_DI|SD_DO|SD_SCK|
//STM32 pin |PA7|PA6|PA5|PA4|PA3|PA2|PA1|PA0| |PB0|PB6|PB7|PB8|PB9| |PA15 |PB5  |PB4  |PB3   | **ALT-SPI1**
//wire color|PRP|BLU|GRN|YEL|ONG|RED|BRN|...
```

Then you can check each wire by its color.   And if you post a photo of your wiring foreign eyes can check the wiring too.

Which country are you in?

David.

madhavanatmel, 2021-09-25:
hi , i have  plugged directly to Arduino UNO and tested this,   but no output in Diagnose sketch . everthing is oxo . looks display is dead . 

the wiring is ok  with stm32 connection .

Regards,
Madhavan - INDIA

prenticedavid, 2021-09-25:
Well,  if it does not work on a Uno it is never going to work on a BluePill.

Follow the advice in the mcufriend_how_to.txt file.
e.g. run LCD_ID_readreg.ino  and copy-paste from Serial Terminal.

David.

madhavanatmel, 2021-09-25:
Read Registers on MCUFRIEND UNO shield

controllers either read Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) F7 F7 F7 F7	Manufacturer ID
reg(0x0009) 08 08 00 00 00	Status Register
reg(0x000A) 08 08	Get Power Mode
reg(0x000C) 66 66	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) FF FF FF FF 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

prenticedavid, 2021-09-25:
Thanks for the readreg report
```
reg(0x0004) F7 F7 F7 F7 Manufacturer ID
...
reg(0x00D3) 00 00 00 00 ILI9341, ILI9488
...
reg(0x00DA) 00 00 RDID1
reg(0x00DB) 00 00 RDID2
reg(0x00DC) 00 00 RDID3
```
So there is no obvious ID reported.   But these registers show that it is a MIPI-style controller :
```
reg(0x0009) 08 08 00 00 00 Status Register
reg(0x000A) 08 08 Get Power Mode
reg(0x000C) 66 66 Get Pixel Format
```

I suggest that you run graphictest_kbv.ino and force `tft.begin(0x9329);` in `setup()`
Please report back with problems e.g. colours, mirroring, scroll directions, readPixel() error, ... 
Please quote what ID is reported on Adafruit Tests page.

David.

madhavanatmel, 2021-09-26:
hi david, goood moring ...  

run the graphictest_kbv.ino and force tft.begin(0x9329); in setup()  .

after that LCD started working with Glue_Demo_320x240 sketechs.

Now LCD Working with UNO & Blue Pill .

tft.begin(0x9329); it's kind of flash reset or chip reset  or flash writing command something like that ?..

prenticedavid, 2021-09-26:
Your new readreg report is different !

```
reg(0x0004) FF FF FF FF Manufacturer ID
```

Although _possible_,  it seems _unlikely_ that 0x9329 would be 100% correct for _every_ test in graphictest_kbv.ino

Likewise,  I would not expect the GLUE examples to work with an unknown ID unless you edit the UTFTGLUE.h file.

If you cooperate properly,  we get your unknown ID controller to work 100% on the Uno.
Then you simply connect the wires to the BluePill and everything works 100% on the BluePill

David.

closed 2021-10-02 by madhavanatmel