prenticedavid/MCUFRIEND_kbv #193: ST7793 DON'T WORKING

kefeur, 2021-12-02:
` Serial.println(g_identifier, HEX);
    //tft.begin(0x9488);
    //tft.begin(0x5393);
    tft.begin(0x7793);//to enable ST7793 driver code`
Im using ST7793 Driver ic for open-smart uno r3 air and 3.5"tft .
This code should control the display, but unfortunately it doesn't work. No reaction, it's all white. What could be the problem ?
I've tried a lot of code, but it's not good.
// GND              -- GND
// 3V3               -- 3.3V
// CS                 -- A3
// RS                 -- A2
// WR                -- A1
// RD                 -- A0
// RST                -- RESET
// LED                -- GND
// DB0                -- 8
// DB1                -- 9
// DB2                -- 10
// DB3                -- 11
// DB4                -- 4
// DB5                -- 13
// DB6                -- 6
// DB7                -- 7
Pinout is correct ? 
Thank's for answer.

prenticedavid, 2021-12-02:
Pinout looks correct.  Note that RST is n.c. on OpenSmart Shields.

```
#elif defined(__AVR_ATmega328P__) && defined(USE_OPENSMART_SHIELD_PINOUT_UNO)
//LCD pins  |D7 |D6 |D5 |D4 |D3 |D2 |D1 |D0 | |RD |WR |RS |CS |RST|
//AVR   pin |PD7|PD6|PB5|PD4|PB3|PB2|PB1|PB0| |PC0|PC1|PC2|PC3|PD2|
//UNO pins  |7  |6  |13 |4  |11 |10 |9  |8  | |A0 |A1 |A2 |A3 |2  |
```

As far as I know the R3 AIR connects the TFT just like the OpenSmart Shields.
So you plug the OpenSmart 3.3V panel into the special socket on the AIR.
You switch to 3.3V logic.

You configure `USE_SPECIAL` and `USE_OPENSMART_SHIELD_PINOUT_UNO` as described in the how_to file

David.

kefeur, 2021-12-03:
Thanks for the reply .
I see the following when I run the test:
TFT LCD test
Using OPENSMART 3.2" TFT Breakout Board Pinout
TFT size is 240x320
Serial took 0ms to start
ID = 0x5393

prenticedavid, 2021-12-03:
I would expect ID to be 0x7793
So the Hi-byte is wrong in two bits i.e. bit5 and bit3 are reading 0.

If you had used Dupont cables I would be suspicious.   But a ready-made 3.3V display and a ready-made AIR board should be reliable.

You can try forcing 0x7793 e.g.
```
    tft.begin(0x7793);
```

I have never tested a ST7793.   SPFD5420, ST7793, ILI9326, R61509V share identical code.

David.

kefeur, 2021-12-03:
Thanks David.
I received a download link from the seller for the product I purchased. It’s practically the things you write in it.
I tested if tft is not connected then I get ID: 0xD3D3. I have two tft modules, they behave exactly the same.

prenticedavid, 2021-12-03:
Looking at the ST7793 datasheet,  there are a few differences between R61509V and ST7793 Manufacturer Registers.

However these do not account for the wrong ID.

What happens if you install the OpenSmart libraries and examples ?

As I said earlier,  I have never tested ST7793.   Someone probably asked and I said try 0xB509 (Renesas R61509V) because the registers were similar.    I presume that they reported success or I would not have added support for 0x7793.

David.

kefeur, 2021-12-03:
The OpenSmart libraries not working.
Do the results of this test help you?

Read Special Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

diagnose any controller
reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 57 97 57 97	Manufacturer ID
reg(0x0009) 08 08 08 08 08	Status Register
reg(0x000A) 08 08	Get Power Mode
reg(0x000C) 5F 9F	Get Pixel Format
reg(0x0030) 20 31 20 31 20	PTLAR
reg(0x0033) 20 20 20 20 20 20 20	VSCRLDEF
reg(0x0061) 20 20	RDID1 HX8347-G
reg(0x0062) 20 20	RDID2 HX8347-G
reg(0x0063) 20 20	RDID3 HX8347-G
reg(0x0064) 24 24	RDID1 HX8347-A
reg(0x0065) 24 24	RDID2 HX8347-A
reg(0x0066) 24 24	RDID3 HX8347-A
reg(0x0067) 24 24	RDID Himax HX8347-A
reg(0x0070) 20 20	Panel Himax HX8347-A
reg(0x00A1) 20 20 20 20 20	RD_DDB SSD1963
reg(0x00B0) 20 20	RGB Interface Signal Control
reg(0x00B3) 20 20 20 20 20	Frame Memory
reg(0x00B4) 24 24	Frame Mode
reg(0x00B6) 24 24 24 24 24	Display Control
reg(0x00B7) 24 24	Entry Mode Set
reg(0x00BF) 2C 2C 2C 2C 2C 2C	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C1) 00 00 00 00	Display Timing
reg(0x00C5) 04 04	Frame Rate
reg(0x00C8) 08 08 08 08 08 08 08 08 08 08 08 08 08	GAMMA
reg(0x00CC) 0C 0C	Panel Control
reg(0x00D0) 00 00 00 00	Power Control
reg(0x00D1) 00 00 00 00	VCOM Control
reg(0x00D2) 00 00 00	Power Normal
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 04 04 04 04	Novatek
reg(0x00DA) 08 08	RDID1
reg(0x00DB) 08 08	RDID2
reg(0x00DC) 0C 0C	RDID3
reg(0x00E0) 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20	GAMMA-P
reg(0x00E1) 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20	GAMMA-N
reg(0x00EF) 2C 2C 2C 2C 2C 2C	ILI9327
reg(0x00F2) 20 20 20 20 20 20 20 20 20 20 20 20	Adjust Control 2
reg(0x00F6) 24 24 24 24	Interface Control
TFT LCD test
Using OPENSMART 3.2" TFT Breakout Board Pinout
TFT size is 240x320
Serial took 0ms to start
ID = 0x5393
Benchmark                Time (microseconds)
Screen fill              TFT LCD test
Using OPENSMART 3.2" TFT Breakout Board Pinout
TFT size is 240x320
Serial took 0ms to start
ID = 0x5393
Benchmark                Time (microseconds)
Screen fill              4846592
Text                     39392
Lines                    1000380
Horiz/Vert Lines         79228
Rectangles (outline)     48588
Rectangles (filled)      1408860
Circles (filled)         622680
Circles (outline)        399052
Triangles (outline)      169560
Triangles (filled)       947540
Rounded rects (outline)  134636
Rounded rects (filled)   1502524

kefeur, 2021-12-03:
I have the solution . I had to replace the libraries. Everything works fine.
Thank you for your patience!

prenticedavid, 2021-12-03:
You must change the defines in LCD_ID_readreg.ino to match your AIR pinout.
Then you will see 
```
diagnose any controller
reg(0x0000) 77 93 ID: ILI9320, ILI9325, ILI9335, ...
...
```

Please tell me **which** libraries worked.   e.g. link to OpenSmart site,   GitHub site, ...

David.

prenticedavid, 2021-12-04:
I can find this item via Google [https://www.aliexpress.com/item/32964204802.html](url)

It looks like your ST7793 + AIR bundle.   There is no link to any software.
I know that Open-Smart have hacked my library.  They have never asked.   I would like to see what they have done.

David.

kefeur, 2021-12-04:
https://l.facebook.com/l.php?u=https%3A%2F%2Fdrive.google.com%2Fdrive%2Ffolders%2F1UDKI61K0t3FdZD_NjZ_A4gWGwfJk4Y3X%3Fspm%3Da2g0s.imconversation.0.0.75ff3e5fMMChvU%26usp%3Dsharing%26fbclid%3DIwAR0aQFBY1N-epz2EJqR2pMXXhqVVWLsqc0dsq5l4fRnR8H09ht0fgmUivmw&h=AT3CjmeZOHVTIvLSuP8a2FihLHp9gichyZr6crgAR-nTZtucXucoXMiv2KrPJ2tm_WDUPx1QjRuYS9M7obOL0sZGV4rksMtihGFbfiDI-5OxqaYHrIX1BtbqdVHSBS-E0zBI
Here's the link to it. I overwrote the original directories with the ones I downloaded from the drive, of course I copied the arduino from the 7793 c/user/documentsarduino/libraries.. folder

kefeur, 2021-12-04:
They may not be happy about it, but I sent it for you. I hope the link works for you too.

prenticedavid, 2021-12-04:
Thanks for the link.   I have had a brief glance at
C:\Users\David Prentice\AppData\Local\Temp\Temp1_[OPEN-SMART] UNO R3 Air + 3.5inch ST7793.zip\[OPEN-SMART] 3.5inch TFT LCD Breakout ST7793\Arduino library\OPENSMART_TFT\MCUFRIEND_kbv.cpp

Note that "Arduino library" folder seems to be duplicated in a nested ZIP.   

I will have a look later this evening.    i.e. to see what hacks they have done and to guess which Release (or Beta) they applied the hack.

David.

kefeur, 2021-12-04:
Your wellcome.
For my part, the problem is solved.
Can i close issue? or no?
Shut up or leave the conversation?

prenticedavid, 2021-12-04:
I would dearly like to resolve 0x7793 in genuine MCUFRIEND_kbv Release v2.99

This would mean:
1.  delete any OpenSmart library.
2.  install MCUFRIEND_kbv **via IDE Library Manager.**
3.  editing utility/mcufriend_shield.h and utility/mcufriend_special.h for USE_SPECIAL and USE_OPENSMART_SHIELD_PINOUT_UNO
4.  running graphictest_kbv.ino from the examples
5.  editing the defines and running LCD_ID_readreg.ino  e.g'
```
//-- Arduino UNO OpenSmart Shield
#define LCD_RST 2  //A4
#define LCD_CS A3
#define LCD_RS A2
#define LCD_WR A1
#define LCD_RD A0

#define LCD_D0 8
#define LCD_D1 9
#define LCD_D2 10 //2
#define LCD_D3 11 //3
#define LCD_D4 4
#define LCD_D5 13 //5
#define LCD_D6 6
#define LCD_D7 7
```

I have looked at the hacked OpenSmart files.   There is nothing substantially different in the 0x7793 code.
They hacked a VERY OLD version of MCUFRIEND_kbv

David.

kefeur, 2021-12-04:
I don't understand what you want due to translation problems?

kefeur, 2021-12-04:
The original pinout is good!
The problem is to be found elsewhere.

kefeur, 2021-12-04:
I noticed that when it didn't work before the library changes, the LED on the D13 didn't flash either.

prenticedavid, 2021-12-04:
I make no secret of the fact that I come from England.  And English is my only language.

It is helpful if GitHub and Arduino.cc members simply quote their country.  It means that others can use "Simplified English".    And Google Translate can put it into your native language.

I gave clear instructions in my last message.   I even numbered each step.
If you do not understand something,  you can just quote the step number.
And then I can explain in more detail.

Life is easier if GitHub members Fork a project.
But you can install a library via Arduino Library Manager and post messages on GitHub or Arduino.cc/Displays forum.

David.

kefeur, 2021-12-04:
I respect your native language.
I don't understand what you want because I don't have an open_smart directory and lib.
I simply copied and overwrote the existing ones.
Adafruit_GFX, OPENSMART_TFT directories in Arduino, and so it works.

kefeur, 2021-12-04:
I can try the things you suggest on another computer.
I don't want it to work on this machine right now. I can only fix this later if it's not a problem.

prenticedavid, 2021-12-04:
Please delete the OPENSMART_TFT directory and install MCUFRIEND_kbv **via the IDE Library Manager.**

I gave numbered instructions.

If the genuine MCUFRIEND_kbv does not work,  you can delete the MCUFRIEND_kbv directory.   Copy the OPENSMART_TFT directory back to your Arduino libraries.

Please say which country.   I can probably introduce you to your neighbour !

David.

kefeur, 2021-12-04:
I don't think the translation is good because I didn't understand what I wanted.
Hungary is my country.
What do I edit in 3 points?

prenticedavid, 2021-12-04:
From C:\Users\ .. \Documents\Arduino\libraries\Mcufriend_kbv\extras\mcufriend_how_to.txt
```

19. OPEN-SMART Shields have different wiring to regular Uno Shields:
    Edit utility/mcufriend_shield.h:  #define USE_SPECIAL
 
19. OPEN-SMART Shields have different wiring to regular Uno Shields:
    Edit utility/mcufriend_shield.h:  #define USE_SPECIAL
    Edit utility/mcufriend_special.h: #define USE_OPENSMART_SHIELD_PINOUT_xxx e.g. USE_OPENSMART_SHIELD_PINOUT_MEGA
    Edit MCUFRIEND_kbv.cpp:           #define SUPPORT_8352B
   Edit utility/mcufriend_special.h: #define USE_OPENSMART_SHIELD_PINOUT_xxx e.g. USE_OPENSMART_SHIELD_PINOUT_MEGA
    Edit MCUFRIEND_kbv.cpp:           #define SUPPORT_8352B
```

In your case,  you only need to 

```
    Edit utility/mcufriend_shield.h:  #define USE_SPECIAL
    Edit utility/mcufriend_special.h: #define USE_OPENSMART_SHIELD_PINOUT_UNO 
```

Yes,  I know that this looks complicated.   But you only do it once.   
All the examples should work with your display.

David.

kefeur, 2021-12-04:
Ok David .
Working for code !
The colors are right.
The display resolution is incorrect

kefeur, 2021-12-04:
[image]
error

kefeur, 2021-12-04:
I did tests and I see that (rotate) orientation 0, orientation1 is good.
orientation2, orientation3 is not good because it shifts the beginning by about 30 pixels.

kefeur, 2021-12-04:
If you ask, I can take a video of how grapictest_kbv works .....

prenticedavid, 2021-12-05:
Please post a video of the "RED" Penguin screen i.e. PORTRAIT_REV

The reason for the 32 rows of garbage is because the controller is 240x432 and your screen is 240x400.
I can solve the shifted pixels for rotation 2 and 3. 
"Vertical Scroll" is always going to have the garbage pixels on a ST7793.

David.

kefeur, 2021-12-05:
Ok I'm making the video.
Why is the top left 0.0?

prenticedavid, 2021-12-05:
0, 0 is just the convention for most controllers.

How are you getting on with the video?
It it is difficult,  just confirm that rotation 0, 1 work 100%.   e.g. "BLUE" penguin and "GREEN" penguin.   Then I can tell you how to fix 2, 3.

David.

kefeur, 2021-12-05:
Gray is not well received by the camera!
The direction of the text is reversed relative to the line .... Examined in another code.
I can't upload either avi or 3gp ... brrrrr

prenticedavid, 2021-12-05:
Do not worry about video.   Your earlier photo is fine.

If you have mirrored text or wrong colours,  just take a photo of the penguin screen.

To fix the shifted picture in rotation 2, 3  edit MCUFRIEND_kbv.cpp 
Look for the following sequence.
```
void MCUFRIEND_kbv::setAddrWindow(int16_t x, int16_t y, int16_t x1, int16_t y1)
{
#if defined(OFFSET_9327)
	if (_lcd_ID == 0x9327) {
	    if (rotation == 2) y += OFFSET_9327, y1 += OFFSET_9327;
	    if (rotation == 3) x += OFFSET_9327, x1 += OFFSET_9327;
    }
#endif

```
change the `if` line to say
```
	if (_lcd_ID == 0x9327 || _lcd_ID == 0x7793) {
```

If photos are difficult.   Just say which penguin and what the problem is.

David.

kefeur, 2021-12-05:
Nothing videos can be uploaded here, very bad!

kefeur, 2021-12-05:
Please provide skype or messenger contact and I'll show you in a live video ...

prenticedavid, 2021-12-05:
People put a video on YouTube.  Then post a link to the site.
Or you could ZIP up a short video clip.  And attach the ZIP file to your GitHub message.

It would be much easier to just edit MCUFRIEND_kbv.cpp and report back in a regular message.  e.g.

PORTRAIT (0 Blue penguin):  ok
LANDSCAPE (1): colours wrong
PORTRAIT_REV (2):  text reversed
LANDSCAPE_REV (3) :  picture shifted 32 pixels.

David.

kefeur, 2021-12-05:
OK.
 tft.begin(0x7793);
 tft.reset();
 tft.setRotation(3);
 tft.fillScreen(BLACK);
 tft.setCursor(0, 0);
 tft.setTextColor(GREEN);
  tft.setTextSize(2);
 tft.print("Zero Test_1");
 tft.fillRect(255, 0, 30, 90,MAGENTA );
small code....
the text is inverted to the box
penguin is ok.
[image]

kefeur, 2021-12-05:
That's good?

prenticedavid, 2021-12-05:
First off.   Report how the graphictest_kbv.ino example was good or bad.

In your "small code"
Omit the `tft.reset(); `statement.   If you use reset() you should call begin() afterwards.

But it is much better to run the library examples.   I know how they should look.

David.

kefeur, 2021-12-05:
It's been 1 hour to upload ....
I didn't say you know what it should look like, it's the other way around
I don’t know what the things in the library should look like, but only 3 work in total. Sd..bmp isn't good either, it doesn't read the sd card.
You will see the video soon I hope.

kefeur, 2021-12-05:
https://youtu.be/U5C0I54juP4

kefeur, 2021-12-05:
[showBMP_kbv_as7_saj1.zip](https://github.com/prenticedavid/MCUFRIEND_kbv/files/7656431/showBMP_kbv_as7_saj1.zip)
My code works, the one in the library is not good. Error message: cannot start SD
SD.begin (SD_CS); repaired SD.begin (5); it doesn't work that way either.

prenticedavid, 2021-12-05:
My apologies.   There are two places in MCUFRIEND_kbv.cpp where you find

```
#if defined(OFFSET_9327)
	if (_lcd_ID == 0x9327) {
```

You need to add 0x7793 in both places.

Regarding SD Card.   Since OpenSmart share the SPI pins you need to make some adjustments on a Uno.    i.e. adding` SPCR = 0 `before writing to TFT and `SPCR = spi_save `before reading from SD Card.

Although Open Smart Shield looks clever it does not work perfectly.
But it is fine for most projects.   You just have to remember the SPCR statements.

David.

kefeur, 2021-12-05:
The 7793 should not be added to the second places because it will spoil the right side of the image, shift it.

prenticedavid, 2021-12-05:
I will have a look at this tomorrow.    I am going to the pub.

The important thing is that you are working 100% in rotation 0 and 1.
It is unlikely that you will actually use "Vertical Scroll" with a ST7793.

David.

David.

kefeur, 2021-12-05:
Have a good rest, have fun.
Thanks for all.

prenticedavid, 2021-12-06:
Please remove the OFFSET_9327 edit(s)

Then search for this sequence:
```
        case 0x5420:
        case 0x7793:
        case 0x9326:
		case 0xB509:
            _MC = 0x200, _MP = 0x201, _MW = 0x202, _SC = 0x210, _EC = 0x211, _SP = 0x212, _EP = 0x213;

```
and change the following 3 lines to:
```
            GS = (val & 0x80) ? (1 << 15)|(4 << 1) : 0; //if GS start at row33 else row1
			uint16_t NL;
			NL = ((400 / 8) - 1) << 9;  // 400 rows

```
This is untested.   But I think it should solve the "shift" problem.
"Vertical Scroll" will always bring garbage into view.

David.

kefeur, 2021-12-06:
All right. i will try.
Please start  i at the beginning.
Should point 0 of tft be at the top left?

kefeur, 2021-12-06:
I would only be able to translate it exactly once ...
All right. i will try.
Please discuss from the beginning.
Is point 0 of tft at the top left?

prenticedavid, 2021-12-06:
Yes,  point 0 is at top left.

Everything is related to (0, 0) i.e. top left in Portrait.
When you do rotations you are changing the controller registers so that (0, 0) is top left in that rotation e.g. Landscape_Rev

Do not worry about these details.   Just make the new edit.   Run graphictest_kbv.ino and report back.

I do not know how well my messages go through Google Translate.
The code text that is in Courier font should not be translated.
I expect you to search for the old sequence.  Then copy-paste the new sequence.

David.

kefeur, 2021-12-06:
I understood what you wanted.
I followed the instructions.
It did not completely solve the problem.
Now my own codes are also displayed incorrectly, I have 1 vertical trash on the right.
Diagnose tft support is good up to the red background, then not, then the blue and green are good again.
graphictest_kbv is good until the blue grades appear, then there's the vertical garbage stripe.
ShowBMP_kbv_uno still doesn't work, "cannot start sd" is the error.
My question was about something else.

prenticedavid, 2021-12-06:
Please describe exactly.  e.g.
graphictest.
Blue penguin.  vertical garbage strip on right.
Green penguin.  horizontal garbage strip at top.
...

diagnose_tft.
blue.  good white rectangles
green.  good white rectangles
red. where are the rectangles?
gray. where are the rectangles?

Please do not ask questions about your own code.
Just run graphictest_kbv and diagnose_TFT_support.
Either describe where the garbage strip is,  the rectangles are wrong.
Or post a photo.

Please remember that I do not have your controller.  Nor do I have your custom sketch.

I will solve your sketch and ShowBMP_kbv_uno **AFTER** I have fixed graphictest_kbv and diagnose_TFT_support.

David.

kefeur, 2021-12-06:
I just described it in detail!
This translator is worth nothing!

prenticedavid, 2021-12-06:
Why not post your messages in Hungarian?
Then I can translate.

kefeur, 2021-12-06:
If you have no experience, you will not succeed because our language is very different.
Ha nincs tapasztalatod , nem fog sikerülni , mert nagyon különnbözik a nyelvünk.

kefeur, 2021-12-06:
Without experience that won't work, our languages are very different.

kefeur, 2021-12-06:
English and Hungarian are very different languages, if you are not familiar with Hungarian it is very easy to mistranslate it.

3 diferent versions !

prenticedavid, 2021-12-06:
Think about it.   You write your message in Hungarian which you are familiar with.

I pass it through Google Translate.   And read the English translation.

Believe me.   I have read Chinese datasheets via Google Translate.   Ok,  some words look strange,  but I can understand what they mean.
"Ha nincs tapasztalatod , nem fog sikerülni , mert nagyon különnbözik a nyelvünk."
is "If you have no experience, you will not succeed because our language is very different."

Please just try it.   This Issue has run for a long time.

David.

kefeur, 2021-12-06:
Rendben de ez legalább 50 fénykép lessz .

prenticedavid, 2021-12-06:
_graphictest_kbv is good until the blue grades appear, then there's the vertical garbage stripe._
One photo of "Blue penguin"

_Diagnose tft support is good up to the red background, then not, then the blue and green are good again._
One photo of red background

That is 2 photos not 50.

David.

kefeur, 2021-12-06:
Szeretnék segíteni a probléma megoldásában .
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

[21 more replies]

closed 2022-02-04 by kefeur