prenticedavid/MCUFRIEND_kbv #195: MCUFRIEND_kbv::vertScroll() only works for orientation 0

beaver700nh, 2021-12-26:
When I run the `scroll_kbv` example unmodified, it works fine but if I change the orientation to 1 instead of 0, it continues to scroll the same direction rather than scrolling in a different way to match the orientation. It would be great if you could change this to support scrolling in orientations other than 0 as well! Perhaps you could implement a `horizScroll()` function.

prenticedavid, 2021-12-26:
The hardware only supports Vertical Scroll.   (And some controllers do not even do that)

When in Landscape the "vertical" means side to side.   As shown in the graphictest_kbv.ino example.

It is much too expensive in time to Horizontal Scroll in software.   The "Software Scroll" is ok for a small block e.g. 200x16.

David.

beaver700nh, 2021-12-27:
Ah, I see. Alright, that's fine then. I decided that I didn't need horizontal scrolling for my project anyways. Thanks for the quick reply!

closed 2021-12-27 by beaver700nh