prenticedavid/MCUFRIEND_kbv #197: 0x5408 text is still reversed as of 1/14/2022

hausofpayne17, 2022-01-14:
I tried the fix in the closed issues

case 0x5408:
        _lcd_capable = 0 | REV_SCREEN | READ_BGR;
        goto common_9320;

but the text is still reversed with latest version of library.

Any recommendations or fixes you can suggest are appreciated.

Thank you for your support of this great library.

prenticedavid, 2022-01-14:
No problem.   I believe that you are working now.

As a general rule.   Always use a "Release".   
Never use the "current Beta" aka. "master" Branch unless specifically asked to do so.

When you use the Arduino IDE Library Manager it installs v2.9.9-Release.

If v2.9.9-Release has a problem it is much easier to describe.  i.e. the Release is a known quantity.

David.

hausofpayne17, 2022-01-14:
Thanks for the last reply David:

I went back to the Library Manager and installed 2.9.8.

I'm running IDE 1.8.19

Still got the reversed text with version 2.9.8

hausofpayne17, 2022-01-14:
Works with 2.9.6. Thank you