prenticedavid/MCUFRIEND_kbv #198: Code stops when calling  TSPoint p = ts.getPoint() from an interrupt routine Arduino UNO with Open Smart LCD

UiARawi, 2022-01-29:
Hi,

First i will thank you for this great work!

I am having an issue when trying to call TSPoint p = ts.getPoint() from TIMER2 ISR in Arduino UNO. 
The code is implementing 5 buttons. Here is the function i am calling from ISR:`
bool Touch_getXY(void)
{
    TSPoint p = ts.getPoint();
    pinMode(YP, OUTPUT);      //restore shared pins
    pinMode(XM, OUTPUT);
    digitalWrite(YP, HIGH);   //because TFT control pins
    digitalWrite(XM, HIGH);
    bool pressed = (p.z > MINPRESSURE && p.z < MAXPRESSURE);
    if (pressed) {
        //LANDSCAPE
        pixel_x = map(p.y, TS_LEFT, TS_RT, 0, tft.width()); 
        pixel_y = map(p.x,TS_BOT , TS_TOP, 0, tft.height());
    }
    return pressed;
}
`

The code is inspired from the button example implemented by .kbv.
The interrupt gets called once every second.

Any help is appreciated!

Regards,
Mohammed

prenticedavid, 2022-01-29:
Please post properly formatted code.  e.g.

```
bool Touch_getXY(void) {
    TSPoint p = ts.getPoint(); 
    pinMode(YP, OUTPUT); //restore shared pins
    pinMode(XM, OUTPUT);
    digitalWrite(YP, HIGH); //because TFT control pins
    digitalWrite(XM, HIGH);
    bool pressed = (p.z > MINPRESSURE && p.z < MAXPRESSURE);
    if (pressed) { //LANDSCAPE
        pixel_x = map(p.y, TS_LEFT, TS_RT, 0, tft.width());
        pixel_y = map(p.x, TS_BOT , TS_TOP, 0, tft.height());
    }
    return pressed;
}
```

I would not advise calling this from an ISR().   It probably works but is "too long and too slow" to call from within an interrupt service routine.

The normal strategy is to set a flag in the ISR().   Poll for this flag in the foreground code.   If set,  execute the function in the foreground.

In practice you tend to just read the current "now" with millis(),  calculate the "next time",   check millis() for "next_time"

David.

UiARawi, 2022-01-30:
Sorry, it is my first time posting an issue on Github!

Thank you for your reply!

I am using a pretty long loop in Arduino (Code size is 31kB), so it takes some time to go through the code. Therefore polling a flag will take as long as the loop gets executed. Is it possible to attach an interrupt to a button?

prenticedavid, 2022-01-30:
Think about what you are trying to achieve.  e.g.

A panic button that needs to be serviced immediately.
A GUI button that can be examined at leisure.
A drawing function that needs to be checked, updated fairly quickly.

Dedicated Touch controllers like XPT2046 will fire an interrupt whenever a Touch is present.
Your unintelligent Touch Panel requires you to monitor it.   And the worst situation of all is when you receive a glitch instead of a positive Touch.

Yes,  you can make periodic checks.   But you need to be careful.
I would just check for a momentary Touch (glitch included).   And set a flag.
Then your program can check properly when it has time to do so.   e.g. make multiple reads,  act on a "debounced" Touch.

But the most important task is to _think_ about your program logic flow.

David.