prenticedavid/MCUFRIEND_kbv #199: about stm32 question

return97, 2022-01-30:
I have a stm32f103c8t6 board

I use the display screen to connect Arduino compilation and run without problem!

But I need to connect the display screen with STM32 and compile it on Arduino ide

What should I do? How to connect? Which library files are modified?

Look forward to your reply!

thank you!

prenticedavid, 2022-01-30:
From mcufriend_kbv.h

```
#elif defined(ARDUINO_GENERIC_STM32F103C) || defined(ARDUINO_BLUEPILL_F103C8) || defined(ARDUINO_BLUEPILL_F103CB)
#warning Uno Shield on BLUEPILL

//LCD pins  |D7 |D6 |D5 |D4 |D3 |D2 |D1 |D0 | |RD |WR |RS |CS |RST| |SD_SS|SD_DI|SD_DO|SD_SCK|
//STM32 pin |PA7|PA6|PA5|PA4|PA3|PA2|PA1|PA0| |PB0|PB6|PB7|PB8|PB9| |PA15 |PB5  |PB4  |PB3   | **ALT-SPI1**

```

Wire according to the "official BLUEPILL" scheme and select BluePill in the IDE.

Most examples should run straight out of the box.

David.

return97, 2022-01-31:
C:\Users\Administrator\Documents\Arduino\libraries\MCUFRIEND_kbv/MCUFRIEND_kbv.h:20:68: error: 'A4' was not declared in this scope
  MCUFRIEND_kbv(int CS=A3, int RS=A2, int WR=A1, int RD=A0, int RST=A4);
                                                                    ^
graphictest_kbv:18:15: error: call to 'MCUFRIEND_kbv::MCUFRIEND_kbv(int, int, int, int, int)' uses the default argument for parameter 1, which is not yet defined
 MCUFRIEND_kbv tft;

this my error!

return97, 2022-01-31:
sorry ! my wrong!!!!  i can now!!!
thanks!!!!
i try it!

return97, 2022-01-31:
By the way, I have another question?

How do I print a variable?

The variables I output are superimposed!

The data before the variable will not be cleared!

prenticedavid, 2022-02-12:
I presume that you have solved this for yourself.

The 5x7 font can be drawn transparently with setTextColor(foreground)
The 5x7 font can be drawn with a fresh background i.e. setTextColor(foreground, background)

The FreeFonts can **only** be drawn transparently with setTextColor(foreground)
You must draw a fresh background manually e.g. with fillRect()

Just write your own helper function for FreeFonts