prenticedavid/MCUFRIEND_kbv #200: Add native support for 0x1581 TFT

mariodantas, 2022-02-06:
Hin is it possible to add native support for 0x1581 TFT ? a kind of define or other to initialize the display ?
Is a 3.5" display

prenticedavid, 2022-02-06:
The Renesas R61581 controller has always been supported by default.
I assume that you have a regular Uno Shield.

Run the diagnose_TFT_support.ino from the examples.   Or just run any of the examples.

If you have a different display,  post a link to the item that **you** have bought.   r.g. Ebay sale page.

mariodantas, 2022-02-06:
[https://www.amazon.fr/gp/product/B091591G6Q/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&th=1](url)

prenticedavid, 2022-02-06:
Your link says

_Conduire IC: ILI9486/ILI9488L_ 

The photo shows a Uno Shield.   I would be much happier if there was a photo of the pcb.

Please plug the Shield into your Uno, Zero, Due, ... as Nature intended.
Run the examples.   Copy-paste any messages from the Serial Terminal.

David.

prenticedavid, 2022-02-13:
Have you identified the controller yet ?
e.g. ILI9486, ILI9488, ILI9481, R61581, ST7796, ...

Amazon, Ebay sellers often lie about their products.
I support most 320x480 controllers.  If you have a new or unusual ID I would be very interested to get it working for you.