prenticedavid/MCUFRIEND_kbv #202: Mcufriend white screen

Yazanghou, 2022-04-10:
Hello programmers!

I have problem with my MCUFriend 2.4" tft, the screen always white and have no display😥

board controler is:
SUM74HC245T
D1409AFP71

And LCD_ID readreg show:
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg[(0x0000) 00 00](tel:0900000000)	ID: ILI9320, ILI9325, ILI9335, ...
reg[(0x0004) 00 00 00 00](tel:09000400000000)	Manufacturer ID
reg[(0x0009](tel:090009)) [00 00 00 00 00](tel:0000000000)	Status Register
reg(0x000A) 00 00	Get Powsr Mode
reg(0x000C) 00 00	Get Pixel Format
reg[(0x0061) 00 00](tel:0900610000)	RDID1 HX8347-G
reg[(0x0062) 00 00](tel:0900620000)	RDID2 HX8347-G
reg[(0x0063) 00 00](tel:0900630000)	RDID3 HX8347-G
reg[(0x0064) 00 00](tel:0900640000)	RDID1 HX8347-A
reg[(0x0065) 00 00](tel:0900650000)	RDID2 HX8347-A
reg[(0x0066) 00 00](tel:0900660000)	RDID3 HX8347-A
reg[(0x0067) 00 00](tel:0900670000)	RDID Himax HX8347-A
reg[(0x0070) 00 00](tel:0900700000)	Panel Himax HX8347-A
reg(0x00A[1) 00 00 00 00 00](tel:10000000000)	RD_DDB SSD1963
reg(0x00B[0) 00 00](tel:00000)	RGB Interface Signal Control
reg(0x00B[4) 00 00](tel:40000)	Inversion Control
reg(0x00B[6) 00 00 00 00 00](tel:60000000000)	Display Control
reg(0x00B[7) 00 00](tel:70000)	Entry Mode Set
reg(0x00BF) [00 00 00 00 00 00](tel:000000000000)	ILI9481, HX8357-B
reg(0x00C[0) 00 21 00](tel:0002100) 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 21 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 A2	Panel Control
reg(0x00D0) 00 21 00	Power Control
reg(0x00D2) 00 A2 A2 A2 A2	NVM Read
reg(0x00D3) 00 A2 A2 A2	ILI9341, ILI9488
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 06 02 05 03 01 47 86 23 03 04 00 00 00 00	GAMMA-P
reg(0x00E[1) 00 00 1](tel:100001)A [20 07 06 05 3](tel:200706053)A [82 40 04 18 07 3](tel:82400418073)F 3F 07	GAMMA-N
reg(0x00EF) [00 00 00 00 00 00](tel:000000000000)	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F[6) 00 00 00 00](tel:600000000)	Interface Control

I have try a lot of solution without and still white!

How to fix this problem?

prenticedavid, 2022-04-10:
I do not recognise this display.
Please post a link to the actual item that you have bought.  e.g. Ebay sale page.

You can try forcing `tft.begin(0x9329)`;

David.

Yazanghou, 2022-04-10:
Thanks
But 0x9329 dosen't work

[image]

This is my tft i bought it from near electroncs saler

Words on ICs:
SUM74HC245T
D1409AFP71

Yazanghou, 2022-04-10:
Also when I run aspect.ino from mcufriends_kpv library I get ID 0xA2A2

prenticedavid, 2022-04-10:
Well,  that looks like a regular Mcufriend Shield.
I would try 0x9320, 0x9325, 0x9329, ...

If none of these show anything,  I would take it back to the local electronics seller.

David.

Yazanghou, 2022-04-10:
Ok I will try....
Thanks 👍🏻

Yazanghou, 2022-04-11:
Finally the problem solved...
I was trying to run it on arduino nano but it didn't work
Now it's working fine on Uno!

Is there a way to run it on nano?

prenticedavid, 2022-04-11:
Please explain.  God gave you a Shield.  God gave you a Uno.
You plug them together as Nature intended.

Please tell me whether the readreg report was from a Uno or from a Nano.
Please tell me which ID works on the Uno.

You can use a Nano with a Shield.   There are Adapters made for a Nano.  i.e. they have sockets for a Nano and sockets for a Shield.

David.

Yazanghou, 2022-04-11:
My project requires small size as possible
The readreg and ID was on nano..
Thank you ❤️ I will search for this adapter

prenticedavid, 2022-04-11:
Please run the readreg sketch on the Uno.
Paste the result here.

Yazanghou, 2022-04-11:
ID now 0x9340

readreg:
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 06	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 02	Inversion Control
reg(0x00B6) 00 0A 82 27 04	Display Control
reg(0x00B7) 00 07	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 21 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 30	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 02 02	NVM Read
reg(0x00D3) 00 00 93 40	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 08 0E 12 05 03 09 47 86 2B 0B 04 00 00 00 00	GAMMA-P
reg(0x00E1) 00 08 1A 20 07 0E 05 3A 8A 40 04 18 0F 3F 3F 0F	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 02 02 02 02 02 02 02 02 02 02 02	Adjust Control 2
reg(0x00F6) 00 01 00 00	Interface Control