prenticedavid/MCUFRIEND_kbv #204: coocox stm32

steevemadison, 2022-05-12:
Hi , i've been using your library with arduino and work like a charm , i got a Coocox Embedded PI v1.0 stm32f103rb , it have 
a form-factor arduino pinout , but got mix GPIO pin , control pin and data pin are mix

A0 = PC0
A1 = PC1
A2 = PC3
A3 = PC3
A4 = PB7
A5 = PB6

D0 = PC11
D1 = PC10
D2 = PC12
D3 = PC6
D4 = PC7
D5 = PC8
D6 = PC9
D7 = PD2
D8 = PA15
D9 = PA8
D10 = PB12
D11 = PB15
D12 = PB14
D13 = PB13
SDA = PB7
SCL = PB6

TFT is 9325 according to arduino core.

i tried to change pin port name in mcufriend_special.h , but nothing work , Touchscreen_STM work, i can read via serial like supposed to,

i never test it with wire , might work , as i want to use the form-factor pinout plugin shield like nucleo

got any recommendation for me ??
thank's in advanced

prenticedavid, 2022-05-12:
Set the appropriate #defines in LCD_ID_readreg.ino sketch.
Verify that it is working e.g. reg(0x0000) is 93 25

Then I will write the appropriate SPECIAL for you.

If it is this board [https://www.cnx-software.com/2013/05/06/coocox-embedded-pi-is-an-stm32-based-mcu-board-that-connects-to-arduino-shields-and-raspberry-pi/](url)
and it is supported by Arduino IDE I could add to the regular "mcufriend_shield.h"

Please confirm the readreg report.
Please confirm the Arduino support.
What is the official -DARDUINO_xxxx macro used by the IDE ?

David.

prenticedavid, 2022-05-12:
The SPECIAL is written.
I don't believe your SDA, SCL pin assignments.  Please confirm them.

steevemadison, 2022-05-12:
yes A2 should be PC2

steevemadison, 2022-05-12:
this what lcd_id gave me 

-Sketch compiled for Arduino UNO:

Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 93 25	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 00 00	Get Power Mode
reg(0x000C) 00 00	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

-Sketch compiled for STM32F103RB Roger Clark Core 0.1.2

Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 02 00 02	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 00 00	Get Power Mode
reg(0x000C) 00 00	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 01	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

prenticedavid, 2022-05-13:
Please use the appropriate defines.   Then you should get the same report with STM32 as you did with the Uno.

Are you really using the "Roger Clark" abortion ?
(It should work but is still a nightmare)

When you post a genuine report,  I will post the (already written) SPECIAL.

David.

steevemadison, 2022-05-13:
the core is from roger clark as generic stm32f103rb , setting 128k flash , 20k sram

the #define for the post above was : 
#define LCD_RST PB7
#define LCD_CS PC3
#define LCD_RS PC2
#define LCD_WR PC1
#define LCD_RD PC0

#define LCD_D0 PA15
#define LCD_D1 PA8
#define LCD_D2 PC12
#define LCD_D3 PC6
#define LCD_D4 PC7
#define LCD_D5 PC8
#define LCD_D6 PC9
#define LCD_D7 PD2

i did a test , from your special.h , i took the wiring from the bluepill section

//LCD pins  |D7  |D6  |D5 |D4 |D3 |D2 |D1  |D0 | |RD |WR |RS |CS |RST| |SD_SS|SD_DI|SD_DO|SD_SCK| |SDA|SCL|
//STM32 pin |PA3 |PA2 |PA1|PA0|PB7|PB6|PA10|PA9| |PB1|PB0|PA7|PA6|PA5| |PB12 |PB15 |PB14 |PB13  | |PB9|PB8|

and rewire my tft shield on a breadboard and i define the pin according to the list and got :

reg(0x0000) 93 25	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 00 00	Get Power Mode
reg(0x000C) 00 00	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

finaly got a reading , but i want to use the form-factor pinout without a mess of wire

thank you

prenticedavid, 2022-05-13:
You have a list of defines.  e.g. from CooCox manual
I have written the SPECIAL for:

```
//LCD pins  |D7 |D6 |D5 |D4 |D3 |D2  |D1 |D0  | |RD |WR |RS |CS |RST| |SD_SS|SD_DI|SD_DO|SD_SCK| |SDA|SCL|
//STM32 pin |PD2|PC9|PC8|PC7|PC6|PC12|PA8|PA15| |PC0|PC1|PC2|PC3|PB7| |PB12 |PB15 |PB14 |PB13  | |PB7|PB6|

```
If you are not prepared to use those defines in LCD_ID_readreg.ino there was no point in me wasting my time and effort.

The CooCox manual shows a proper Shield socket.   So as soon as LCD_ID_readreg has been tested,  I post the SPECIAL.   You plug the Shield into the CooCox.  Enable the SPECIAL.   All the TFTs should work.

David.

steevemadison, 2022-05-14:
thank's for your time , i went back to my initial setup , TFT is mount in the coocox header , no wire .
define my pinout like this in LCD_ID : 

#define LCD_RST PB7
#define LCD_CS PC3
#define LCD_RS PC2
#define LCD_WR PC1
#define LCD_RD PC0

#define LCD_D0 PA15
#define LCD_D1 PA8
#define LCD_D2 PC12
#define LCD_D3 PC6
#define LCD_D4 PC7
#define LCD_D5 PC8
#define LCD_D6 PC9
#define LCD_D7 PD2

Arduino IDE v1.8.19
Roger Clark STM32 Core v0.1.2
Generic STM32F103R board
variant STM32F103RB 20k RAM , 128k flash
upload method serial ( i'm using FTDI)
72Mhz CPU speed
Optimize Smallest default

first try with default serial , i got this : 
reg(0x0000) 93 25	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 00 00	Get Power Mode
reg(0x000C) 00 00	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

second try with serial3 instead

reg(0x0000) 93 25	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 00 00	Get Power Mode
reg(0x000C) 00 00	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 93 25 93 25 93 25 93 25 93	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

prenticedavid, 2022-05-14:
Add this define to mcufriend_special.h
```
#define USE_COOCOX_STM32
```
with this switch block
```
//################################### COOCOX_STM32 on STM32103RB ##############################
#elif defined(USE_COOCOX_STM32) && (defined(ARDUINO_GENERIC_STM32F103R)||defined(ARDUINO_GENERIC_F103RBTX))
#warning Uno Shield on USE_COOCOX_STM32

//LCD pins  |D7 |D6 |D5 |D4 |D3 |D2  |D1 |D0  | |RD |WR |RS |CS |RST| |SD_SS|SD_DI|SD_DO|SD_SCK| |SDA|SCL|
//STM32 pin |PD2|PC9|PC8|PC7|PC6|PC12|PA8|PA15| |PC0|PC1|PC2|PC3|PB7| |PB12 |PB15 |PB14 |PB13  | |PB7|PB6|

#if defined(ARDUINO_GENERIC_F103RBTX)   //regular CMSIS libraries
#define REGS(x) x
#define GPIO_INIT()   { RCC->APB2ENR |= RCC_APB2ENR_IOPAEN | RCC_APB2ENR_IOPBEN | RCC_APB2ENR_IOPCEN | RCC_APB2ENR_IOPDEN | RCC_APB2ENR_AFIOEN; \
        AFIO->MAPR |= AFIO_MAPR_SWJ_CFG_1;}
#else                                                                  //weird Maple libraries
#define REGS(x) regs->x
#endif

#define WRITE_DELAY { }
#define READ_DELAY  { RD_ACTIVE4; }
#define GROUP_MODE(port, reg, mask, val)  {port->REGS(reg) = (port->REGS(reg) & ~(mask)) | ((mask)&(val)); }
#define GP_OUT(port, reg, mask)           GROUP_MODE(port, reg, mask, 0x33333333)
#define GP_INP(port, reg, mask)           GROUP_MODE(port, reg, mask, 0x44444444)
#define PIN_OUTPUT(port, pin) {\
        if (pin < 8) {GP_OUT(port, CRL, 0xF<<((pin)<<2));} \
        else {GP_OUT(port, CRH, 0xF<<((pin&7)<<2));} \
    }
#define PIN_INPUT(port, pin) { \
        if (pin < 8) { GP_INP(port, CRL, 0xF<<((pin)<<2)); } \
        else { GP_INP(port, CRH, 0xF<<((pin&7)<<2)); } \
    }
#define PIN_HIGH(port, pin)   (port)-> REGS(BSRR) = (1<<(pin))
#define PIN_LOW(port, pin)    (port)-> REGS(BSRR) = (1<<((pin)+16))

#define RD_PORT GPIOC
#define RD_PIN  0
#define WR_PORT GPIOC
#define WR_PIN  1
#define CD_PORT GPIOC
#define CD_PIN  2
#define CS_PORT GPIOC
#define CS_PIN  3
#define RESET_PORT GPIOB
#define RESET_PIN  7

// configure macros for the data pins
#define AMASK ((1<<8)|(1<<15))
#define CMASK ((15<<6)|(1<<12))
#define DMASK (1<<2)
#define write_8(d)    { GPIOA->REGS(BSRR) = AMASK << 16; GPIOC->REGS(BSRR) = CMASK << 16;\
                        GPIOD->REGS(BSRR) = DMASK << 16; \
                        GPIOA->REGS(BSRR) = (((d) & (1<<0)) << 15); \
                        GPIOA->REGS(BSRR) = (((d) & (1<<1)) << 4); \
                        GPIOC->REGS(BSRR) = (((d) & (1<<2)) << 10); \
                        GPIOC->REGS(BSRR) = (((d) &(15<<3)) << 3); \
                        GPIOD->REGS(BSRR) = (((d) & (1<<7)) >> 5); \
                      }
#define read_8()     ( ((GPIOA->REGS(IDR) & (1<<15)) >> 15) \
                     | ((GPIOA->REGS(IDR) & (1<<8))  >> 7) \
                     | ((GPIOC->REGS(IDR) & (1<<12)) >> 10) \
                     | ((GPIOC->REGS(IDR) & (15<<6)) >> 3) \
                     | ((GPIOD->REGS(IDR) & (1<<2))  << 5) \
                     )
//                                          PA15,PA8                 PC12,PC9-PC8                         PC7,PC6                        PD2
#define setWriteDir() {GP_OUT(GPIOA, CRH, 0xF000000F); GP_OUT(GPIOC, CRH, 0xF00FF); GP_OUT(GPIOC, CRL, 0xFF000000); GP_OUT(GPIOD, CRL, 0xF00); }
#define setReadDir()  {GP_INP(GPIOA, CRH, 0xF000000F); GP_INP(GPIOC, CRH, 0xF00FF); GP_INP(GPIOC, CRL, 0xFF000000); GP_INP(GPIOD, CRL, 0xF00); }

#define write8(x)     { write_8(x); WRITE_DELAY; WR_STROBE; }
#define write16(x)    { uint8_t h = (x)>>8, l = x; write8(h); write8(l); }
#define READ_8(dst)   { RD_STROBE; READ_DELAY; dst = read_8(); RD_IDLE; }
#define READ_16(dst)  { uint8_t hi; READ_8(hi); READ_8(dst); dst |= (hi << 8); }

```

steevemadison, 2022-05-15:
thank you , i have tried it , upload the graphictest to my coocox , open the serial and i get : 

Serial took 0ms to start
ID = 0x9325

and the screen stay white , 
if i don't put this line juste after the setup: 

void setup(void) {
   afio_cfg_debug_ports(AFIO_DEBUG_NONE);
  
i get from the serial : 

Serial took 0ms to start
ID = 0x0

i've search to try to find something , didn't find anything.

prenticedavid, 2022-05-15:
First off.   Why are you using the weird RogerClark core ?
Especially since the Official STM32 Core is written and maintained by STMicroelectronics who actually made your chip.

I only have an ancient 2017 version on my PC.  After all there is no proper Release procedure from the "RogerClark" ftraternity.   And I would not trust them anyway !!

You may have noticed that the regular GPIO_INIT() macro says:
```
#define GPIO_INIT()   { RCC->APB2ENR |= RCC_APB2ENR_IOPAEN | RCC_APB2ENR_IOPBEN | RCC_APB2ENR_IOPCEN | RCC_APB2ENR_IOPDEN | RCC_APB2ENR_AFIOEN; \
        AFIO->MAPR |= AFIO_MAPR_SWJ_CFG_1;}
```
The 2017 RogerClark sets the AFIO->MAPR and RCC registers in init().

From memory the MAPR setting enables PB4 as GPIO.   You are not using PB4.

[image]

You are using PA15.   I guess that you never use JTAG.  You use SWD like everyone else.

I really don't want to trawl through RogerClark abortion documents.   That is your job.

Please let me know how you get on.   You will be adjusting the GPIO_INIT() macro.
Yes,   it will me useful to know for the civilised world.

David.

prenticedavid, 2022-05-21:
Bump.  Have you made any progress ?

Unless you have a very good reason for using the weird RogerClark core,  I suggest that you use the official STM32 Core.

If you really **must** use the weird RogerClark core,  please quote the **actual** version number.
Otherwise I can't replicate anything.

David.