prenticedavid/MCUFRIEND_kbv #205: Warning spam

beaver700nh, 2022-06-19:
Hi

Every time I try to clean build my project, I am flooded with these warnings:
```
/home/pi/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp: In member function 'uint16_t MCUFRIEND_kbv::readReg(uint16_t, int8_t)':
/home/pi/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:162:13: warning: unused variable 'lo' [-Wunused-variable]
     uint8_t lo;
             ^~
/home/pi/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp: In member function 'uint16_t MCUFRIEND_kbv::readID()':
/home/pi/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:258:5: warning: this 'if' clause does not guard... [-Wmisleading-indentation]
     if (ret == 0x1526)          //R61526 [xx 06 15 26] if I have written NVM
     ^~
/home/pi/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:260:2: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
  if (ret == 0x89F0)          //ST7735S: [xx 7C 89 F0]
  ^~
/home/pi/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:194:19: warning: unused variable 'ret2' [-Wunused-variable]
     uint16_t ret, ret2;
                   ^~~~
/home/pi/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp: In member function 'int16_t MCUFRIEND_kbv::readGRAM(int16_t, int16_t, uint16_t*, int16_t, int16_t)':
/home/pi/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:294:22: warning: unused variable 'tmp' [-Wunused-variable]
     uint8_t r, g, b, tmp;
                      ^~~
/home/pi/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp: In member function 'virtual void MCUFRIEND_kbv::setRotation(uint8_t)':
/home/pi/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:355:29: warning: unused variable 'REV' [-Wunused-variable]
     uint16_t GS, SS_v, ORG, REV = _lcd_rev;
                             ^~~
/home/pi/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp: In member function 'void MCUFRIEND_kbv::begin(uint16_t)':
/home/pi/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:1584:31: warning: unused variable 'ST7781_regValues' [-Wunused-variable]
         static const uint16_t ST7781_regValues[] PROGMEM = {
                               ^~~~~~~~~~~~~~~~
/home/pi/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:1684:30: warning: unused variable 'ST7789_regValues_arcain6' [-Wunused-variable]
         static const uint8_t ST7789_regValues_arcain6[] PROGMEM = {
                              ^~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:2595:30: warning: unused variable 'ILI9341_regValues_ada' [-Wunused-variable]
         static const uint8_t ILI9341_regValues_ada[] PROGMEM = {        // Adafruit_TFTLCD only works with EXTC=0
                              ^~~~~~~~~~~~~~~~~~~~~
/home/pi/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:2675:30: warning: unused variable 'ILI9481_CPT29_regValues' [-Wunused-variable]
         static const uint8_t ILI9481_CPT29_regValues[] PROGMEM = {    // 320x430
                              ^~~~~~~~~~~~~~~~~~~~~~~
/home/pi/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:2687:30: warning: unused variable 'ILI9481_PVI35_regValues' [-Wunused-variable]
         static const uint8_t ILI9481_PVI35_regValues[] PROGMEM = {    // 320x480
                              ^~~~~~~~~~~~~~~~~~~~~~~
/home/pi/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:2699:30: warning: unused variable 'ILI9481_AUO317_regValues' [-Wunused-variable]
         static const uint8_t ILI9481_AUO317_regValues[] PROGMEM = {    // 320x480
                              ^~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:2710:30: warning: unused variable 'ILI9481_CMO35_regValues' [-Wunused-variable]
         static const uint8_t ILI9481_CMO35_regValues[] PROGMEM = {    // 320480
                              ^~~~~~~~~~~~~~~~~~~~~~~
/home/pi/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:2724:30: warning: unused variable 'ILI9481_RGB_regValues' [-Wunused-variable]
         static const uint8_t ILI9481_RGB_regValues[] PROGMEM = {    // 320x480
                              ^~~~~~~~~~~~~~~~~~~~~
/home/pi/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:2816:7: warning: label 'common_9488' defined but not used [-Wunused-label]
       common_9488:
       ^~~~~~~~~~~
```

It is rather annoying, and it would be great if these could be patched. Thanks!

prenticedavid, 2022-06-19:
No,  I have zero intention to remove those unused variable warnings.

v2.9.9 is two years old.  Yes,  it is about time that I issue a new Release.   I have been saying this for at least the last year.

beaver700nh, 2022-06-19:
Hi, so is there anything that will be done about it? And what and when will be the new release?