prenticedavid/MCUFRIEND_kbv #207: tft.readID() finds: ID = 0xC6C6 This ID is not supported

the-indian-dev, 2022-06-29:
Hi,
I tried to run the ``Glue_Demo_480x320`` example and the other examples but I see a white blank screen, so I ran the Diagnostic example code, which says my TFT is not supported.
I got this TFT Display from a local shop.
The Box of the display says: ``3.5" UNO Module Driver: ILI9486 Bus:8Bit Dots: 320x480 Touch: Yes``

Output of LCD_ID_readreg :

```
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 40 40	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 40 54 C0 66	Manufacturer ID
reg(0x0009) 40 40 61 40 40	Status Register
reg(0x000A) 40 48	Get Power Mode
reg(0x000C) 40 66	Get Pixel Format
reg(0x0061) 40 40	RDID1 HX8347-G
reg(0x0062) 40 40	RDID2 HX8347-G
reg(0x0063) 40 40	RDID3 HX8347-G
reg(0x0064) 40 40	RDID1 HX8347-A
reg(0x0065) 40 40	RDID2 HX8347-A
reg(0x0066) 40 40	RDID3 HX8347-A
reg(0x0067) 40 40	RDID Himax HX8347-A
reg(0x0070) 40 40	Panel Himax HX8347-A
reg(0x00A1) 40 D3 70 D3 70	RD_DDB SSD1963
reg(0x00B0) 40 40	RGB Interface Signal Control
reg(0x00B4) 40 40	Inversion Control
reg(0x00B6) 40 42 42 7B 7B	Display Control
reg(0x00B7) 40 46	Entry Mode Set
reg(0x00BF) 40 40 40 40 40 40	ILI9481, HX8357-B
reg(0x00C0) 40 4E 4E 4E 4E 4E 4E 4E 4E	Panel Control
reg(0x00C8) 40 40 40 40 40 40 40 40 40 40 40 40 40	GAMMA
reg(0x00CC) 40 44	Panel Control
reg(0x00D0) 40 40 40	Power Control
reg(0x00D2) 40 40 40 40 40	NVM Read
reg(0x00D3) 40 40 D4 C6	ILI9341, ILI9488
reg(0x00D4) 40 40 40 40	Novatek ID
reg(0x00DA) 40 54	RDID1
reg(0x00DB) 40 C0	RDID2
reg(0x00DC) 40 66	RDID3
reg(0x00E0) 40 46 5D 4B 4F 5F 43 7F C6 71 4F 5A 45 7E 6B 4D	GAMMA-P
reg(0x00E1) 40 4A 7B 6E 49 5D 4B 6A C3 5B 47 5F 47 62 01 41	GAMMA-N
reg(0x00EF) 40 80 40 10 60 40	ILI9327
reg(0x00F2) 40 18 A3 12 02 B2 12 FF 10 40 40 40	Adjust Control 2
reg(0x00F6) 00 54 80 66	Interface Control
```
Please help me get this LCD working!

prenticedavid, 2022-06-29:
You may have bought it in a local shop but you can find the display online.  e.g. Ebay, AliExpress, 
Compare the item on your desk with the photos on Ebay sale page.
If you can identify the exact same item post the link with a note to say "looks like my item"

The readreg output clearly shows bit#6 is stuck high (0x40).
If you remove bit#6 you would see
```

reg(0x00D3) 40 40 D4 C6	ILI9341, ILI9488
reg(0x00D3) 00 00 94 86 ILI9341, ILI9488

```

I suggest that you just force the ID e.g.`   tft.begin(0x9486);`

Hint.  The Touch screen is often on LCD_D7, LCD_D6 pins.
If the Touch is broken or damaged it might make LCD_D6 to always "read high"

David.

the-indian-dev, 2022-06-29:
I forced the example ``TouchScreen_Callibr_Native`` to run with 0x9486 ID however, now it shows ``Waiting for Serial`` mirrored and goes to a white blank screen after 2-3 seconds.

I wasn't able to find any exact match for the display. It looks like most of the 3.5" Resistive Touch Screen Blue coloured ILI9486 displays which come with a stylus.

``If you remove bit#6 you would see``
I am new to programming TFT displays. What does it mean?

the-indian-dev, 2022-06-29:
``If you find a "similar" display online I want to see the link. NOT words``
Looks like this one but the font on the back is different
https://www.electronicscomp.com/3.5-inch-tft-lcd-touch-screen-display-shield-for-arduino-uno

``diagnose_controller``
Gives correct results, no weird colour stuff or anything.
``graphictest_kbv``
Correct Results. I ran the program 3 times. One time it messed up everything. Shapes get drawn in random places as if the coordinates are wrong. I wasn't able to reproduce the issue again.

I think this display has the chip soldered loosely, The examples work sometimes or other times the display just completely messes up.

closed 2022-06-29 by the-indian-dev