prenticedavid/MCUFRIEND_kbv #209: Working with ESP32 pins greater than 31

FBSeletronica, 2022-07-15:
I'm working on a project with ESP32-S2 and I noticed that the library doesn't allow using pins greater than 31.

Looking at the library, I noticed that the registers of these pins are not mapped. 

Would it be possible to add support for ESP32-S2 to use all pins?

prenticedavid, 2022-07-16:
There is a ready-made ESP32 board with Arduino headers i.e. fromTTGO.
Hence the "official shield" wiring in mcufriend_shield.h

If you can find a read-made ESP32-S2 board with Arduino headers I will buy one.  And add the appropriate entry to mcufriend_shield.h

Otherwise you are at liberty to write whatever custom SPECIAL scheme you like to mcufriend_special.h
I have never used (or seen) an ESP32-S2 board.   I guess that all of the user pins are supported by pinMode() and digitalWrite()

Which TFT controller are you using?
Bodmer supports several "popular" 8-bit parallel controllers with TFT_eSPI library.

David.

prenticedavid, 2022-07-23:
Bump.

If there is an Arduino header S2 board I will be happy to buy one.  And support it in mcufriend_shield.h

David.