prenticedavid/MCUFRIEND_kbv #210: Whitescreen glitches when calling a simple print

Trahor, 2022-08-15:
Hey there, beginner here.

I've encountered some weird white screen glitches only when I'm calling the print on a specific loop, which is quite sad because I wanted to display what numbers I've just pressed.
But instead of actually printing that number, I've been "stuck" on a white screen even tho the program was still running behind (100% sure, serial outputs were there) and was supposed to fillScreen(BLACK) at the end of my function. After some troubleshoot I've decided to go back to that stuck whitescreen state but it just didn't behaved the same (just doing nothing), and now I just got white flashing screen everytime I press a button. 

I've checked if my cursor wasn't getting out of bounds for some reason but nothing seems out of place as far as I can tell.
Everything is working everywhere else, none of the other print are broken.

So I'm kinda lost because a print shouldn't do this and I hope I haven't make a mistake and wasted your time.
I've cut off the part of my code which is having this issue as best as I could so you can check it out, I've also added some comments so you can understand where it got messy.

Also I should probably tell you what I'm using, so I have an Arduino Mega with this kind of screen https://www.amazon.fr/gp/product/B07PLQS8MT/ref=ppx_yo_dt_b_asin_title_o04_s01?ie=UTF8&psc=1

Please be kind with my code, I really have no solid knowledge about programing and I'm just tinkering on my own, discovering is much more fun than learning ^^
I really hope I didn't make a huge mistake anybody could have seen, I'd feel really dumb.

[tft issue.txt](https://github.com/prenticedavid/MCUFRIEND_kbv/files/9338957/tft.issue.txt)

prenticedavid, 2022-08-15:
I always advise running the examples before writing any custom code.  
You should always restore pinMode **after** a `ts.getPoint()` call
e.g. from button_simple.ino
```
bool Touch_getXY(void)
{
    TSPoint p = ts.getPoint();
    pinMode(YP, OUTPUT);      //restore shared pins
    pinMode(XM, OUTPUT);
    ...
```

Trahor, 2022-08-15:
Hey, thanks for the quick answer.
As I expected I feel now super dumb. I did built everything from your examples, your codes are very clear so it was "easy" to understand even for me. I remember seeing what you quote, but I don't know when did I move that part and why...

And now that I think of it, I know why it does still work for the other part of my code, I'm calling another function which is written correctly with the pinMode **after** the call
Now indeed everything is working fine.

Thanks a lot and sorry for the inconvenience !

closed 2022-08-15 by Trahor