prenticedavid/MCUFRIEND_kbv #212: Adafruit LoRa32u4 Pins

McrESP, 2022-08-22:
I'm using ATmega a 3.5" touch screen and a Adafruit  LoRa radio as a remote gate and kennel opener. It works like a charm so thank you for providing all the tools needed to make it happen.
But now I need to make it much more compact and portable by using a Adafruit LoRa32u4 and a battery backup.
Here are the pins I'm using:
// adapted from LCD_ID_Reader from http://misc.ws/lcd_information
// controllers either read as 16-bit or as a sequence of 8-bit values

//-- Arduino UNO or Mega 2560 Plugged as shield
#define LCD_RST A4
#define LCD_CS A3
#define LCD_RS A2
#define LCD_WR A1
#define LCD_RD A0

#define LCD_D0 11
#define LCD_D1 12
#define LCD_D2 2
#define LCD_D3 3
#define LCD_D4 5
#define LCD_D5 6
#define LCD_D6 9
#define LCD_D7 10

And here are the results from LDC_ID_readnew:

Read Special Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

diagnose any controller
reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 06	Get Pixel Format
reg(0x0030) 00 00 00 00 00	PTLAR
reg(0x0033) 00 00 00 00 00 00 00	VSCRLDEF
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B3) 00 00 00 00 00	Frame Memory
reg(0x00B4) 00 00	Frame Mode
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 21 00 00 00 00 00 00 00	Panel Control
reg(0x00C1) 00 10 10 10	Display Timing
reg(0x00C5) 00 31	Frame Rate
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 71	Panel Control
reg(0x00D0) 00 21 00 00	Power Control
reg(0x00D1) 00 10 10 10	VCOM Control
reg(0x00D2) 00 00 00	Power Normal
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek
reg(0x00DA) 00 C3	RDID1
reg(0x00DB) 00 39	RDID2
reg(0x00DC) 00 71	RDID3
reg(0x00E0) 00 0F 16 14 0A 0D 06 43 75 33 06 0E 00 0C 09 08	GAMMA-P
reg(0x00E1) 00 08 2B 2D 04 10 04 3E 24 4E 04 0F 0E 35 38 0F	GAMMA-N
reg(0x00EF) 00 03 80 02 02 02	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

Do these results look like the wiring is correct and if so is there a special for the 32u4?

Thanks for any help

prenticedavid, 2022-08-22:
Your first regs look normal.  But 0xDA, DB, DC look wrong.   Are you sure that your wiring is correct ?

What ID is returned when you plug the Shield into a Uno ?
Run LCD_ID_readreg.ino on the Uno.   Compare the result of your custom wiring to the "known" result from the Shield + Uno.

David.

McrESP, 2022-08-22:
David,
Thanks for getting back to me so soon.
I'm doing something wrong for sure.
I double checked the wiring and it looked good.
I don't have an UNO so I tested it on a MEGA2560 and the results are way different.

With 32u4LoRa

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 06	Get Pixel Format
reg(0x0030) 00 00 00 00 00	PTLAR
reg(0x0033) 00 00 00 00 00 00 00	VSCRLDEF
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B3) 00 00 00 00 00	Frame Memory
reg(0x00B4) 00 00	Frame Mode
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 21 00 00 00 00 00 00 00	Panel Control
reg(0x00C1) 00 10 10 10	Display Timing
reg(0x00C5) 00 31	Frame Rate
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 71	Panel Control
reg(0x00D0) 00 21 00 00	Power Control
reg(0x00D1) 00 10 10 10	VCOM Control
reg(0x00D2) 00 00 00	Power Normal
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek
reg(0x00DA) 00 C3	RDID1
reg(0x00DB) 00 39	RDID2
reg(0x00DC) 00 71	RDID3
reg(0x00E0) 00 0F 16 14 0A 0D 06 43 75 33 06 0E 00 0C 09 08	GAMMA-P
reg(0x00E1) 00 08 2B 2D 04 10 04 3E 24 4E 04 0F 0E 35 38 0F	GAMMA-N
reg(0x00EF) 00 03 80 02 02 02	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

With MEGA256

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 01 01 01 01 01	Status Register
reg(0x000A) 02 02	Get Power Mode
reg(0x000C) 00 44	Get Pixel Format
reg(0x0061) 01 01	RDID1 HX8347-G
reg(0x0062) 02 02	RDID2 HX8347-G
reg(0x0063) 03 03	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 01 01	RDID2 HX8347-A
reg(0x0066) 02 02	RDID3 HX8347-A
reg(0x0067) 03 03	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 81 81 81 81 81	RD_DDB SSD1963
reg(0x00B0) 80 80	RGB Interface Signal Control
reg(0x00B4) 80 80	Inversion Control
reg(0x00B6) 82 82 82 82 82	Display Control
reg(0x00B7) 83 83	Entry Mode Set
reg(0x00BF) 83 83 83 83 83 83	ILI9481, HX8357-B
reg(0x00C0) 80 80 80 80 80 80 80 80 80	Panel Control
reg(0x00C8) 80 88 80 80 80 80 80 80 80 80 80 80 80	GAMMA
reg(0x00CC) 80 80	Panel Control
reg(0x00D0) 80 80 80	Power Control
reg(0x00D2) 82 82 82 82 82	NVM Read
reg(0x00D3) 83 83 83 83	ILI9341, ILI9488
reg(0x00D4) 80 80 80 80	Novatek ID
reg(0x00DA) 82 82	RDID1
reg(0x00DB) 83 83	RDID2
reg(0x00DC) 80 98	RDID3
reg(0x00E0) 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80 80	GAMMA-P
reg(0x00E1) 81 81 81 81 81 81 81 81 81 81 81 81 81 81 81 81	GAMMA-N
reg(0x00EF) 83 83 83 83 83 83	ILI9327
reg(0x00F2) 82 82 82 82 82 82 82 82 82 82 82 82	Adjust Control 2
reg(0x00F6) 82 82 82 82	Interface Control

Any suggestions?

prenticedavid, 2022-08-22:
I don't believe you.   Have you plugged the Shield into the Mega2560 as Nature intended?

Have you used the defines that come with the library installation ?

David.

McrESP, 2022-08-22:
You are absolutely right.
I forgot to change to pins back to the MEGA2560
here is the serial output with the correct pins.

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 06	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 02	Inversion Control
reg(0x00B6) 00 0A 82 27 04	Display Control
reg(0x00B7) 00 06	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 21 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 71	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 03 03	NVM Read
reg(0x00D3) 00 00 93 41	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 0F 16 14 0A 0D 06 43 75 33 06 0E 00 0C 09 08	GAMMA-P
reg(0x00E1) 00 08 2B 2D 04 10 04 3E 24 4E 04 0F 0E 35 38 0F	GAMMA-N
reg(0x00EF) 00 03 80 02 02 02	ILI9327
reg(0x00F2) 00 02 02 02 02 02 02 02 02 02 02 02	Adjust Control 2
reg(0x00F6) 00 01 00 00	Interface Control

prenticedavid, 2022-08-22:
There you are.  A regular ILI9341.  Which is 240x320 and very _unlikely_ to be 3.5 inch diagonal.
```
reg(0x00D3) 00 00 93 41 ILI9341, ILI9488
```

Anyway,  check your 32u4 defines.   Run the sketch.   When you get the **same** report as Mega2560 paste your defines.

I will write the appropriate SPECIAL for you.

David.

McrESP, 2022-08-22:
You are correct, it's a 2.8" that i was testing with.
Switching to the 3.5" now.

McrESP, 2022-08-23:
David,
Thanks for all the help with my project but I really screwed this up. 
I forgot about the logic levels on the 32u4 being 3.3v and the LCD levels being 5v.
I only have 1, 8 bit logic level converter so I'm putting the project on hold till I receive another.
Thanks again and everything works great on the MEGA.