prenticedavid/MCUFRIEND_kbv #213: Cannot interface Due + elechouse tft Shield V2.1 + TFT_320QDT_9341, with Boiana_EQ code

SyberStark, 2022-08-25:
All I want is to run Boiana_Eq code with DUE + Elechouse TFT shield for Due V2.1 + TFT_320QDT_ILI9341 - 320x240 screen.
Boiana_EQ code: https://drive.google.com/drive/folders/1no3Nnc6ZwLjqzFxb2D6PBbd0kwbAyVLU?usp=sharing

**TFT_320QDT_ILI9341 320x240 :**
[image]

**Elechouse TFT shield for Due V2.1:**
[image]

_____________________________________________________________________________________________________________________________________________
The TFT_320QDT_ILI9341 - 320x240 screen **DOES NOT** work with UTFT library connected with DUE along with Elechouse shield (TFT shield for due) V2.1.
UTFT code:
https://drive.google.com/file/d/1r70SulprEElM27l-s8u7E_6aSPXTeXGT/view?usp=sharing
Note: I have changed HW_ARM_defines.h ( uncommented #define EHOUSE_DUE_SHIELD 1 ) where as //#define CTE_DUE_SHIELD 1 is still commented. 

_____________________________________________________________________________________________________________________________________________
The TFT_320QDT_ILI9341 - 320x240 screen, WORKS well with UTFT library connected with Mega2560 along with TFT shield of mega(V 2.2).
(Hence confirming that my screen is not faulty)

 **TFT shield of mega(V 2.2) + Mega 2560 :**
[image]

prenticedavid, 2022-08-27:
There are 3 styles of Shield pinout.
1.  data bus on D22-D37 e.g. Mega2560
2.  data bus on ARM PCxx pins CTE_DUE
3.  data bus on different ARM PCxx pins EHOUSE_DUE

I am the author of MCUFRIEND_kbv not UTFT.

For UTFT you should edit  C:\Users\David\Documents\Arduino\libraries\UTFT\hardware\arm\HW_ARM_defines.h

```
// ElecHouse TFT LCD/SD Shield for Arduino Due
// -------------------------------------
// Uncomment the following line if you are using this shield
#define EHOUSE_DUE_SHIELD 1
//
// For this shield: RS=22, WR=23, CS=31, RST=33
//********************************************************************
```

I would expect the Display to work with the correct constructor e.g.
```

// ElecHouse TFT LCD/SD Shield for Arduino Due : <display model>,22,23,31,33
//
// Remember to change the model parameter to suit your display module!
UTFT myGLCD(ILI9341_16,22,23,31,33);
```

I do not have the EHOUSE Adapter.   PLease let me know how you get on.

When it is working with UTFT please test the appropriate SPECIAL for MCUFRIEND_kbv
I only have the CTE_DUE adapter.

David.

SyberStark, 2022-08-29:
Dear David. Thankx for your reply. 

### **1.) UTFT + ELECHOUSE due shield + ILI9341 screen + DUE**

> I did exactly what you guided, changed the **HW_ARM_defines.h**
> This is [ELECHOUSE ADAPTER](https://www.elechouse.com/elechouse/images/product/3.2TFT%20shield/Pin%20Mapping.pdf) for DUE 

After change:
```
// CTE TFT LCD/SD Shield for Arduino Due
// -------------------------------------
// Uncomment the following line if you are using this shield
//#define CTE_DUE_SHIELD 1
//
// For this shield: RS=25, WR=26, CS=27, RST=28
//********************************************************************

// ElecHouse TFT LCD/SD Shield for Arduino Due
// -------------------------------------
// Uncomment the following line if you are using this shield
#define EHOUSE_DUE_SHIELD 1
//
// For this shield: RS=22, WR=23, CS=31, RST=33-
//********************************************************************
```
After making these changes, I ran Arduino IDE \File \examples\UTFT\Arduino(ARM)+Teensy\UTFT_Demo_320x240
And used `UTFT myGLCD(ILI9341_16,22,23,31,33);` as a constructor. 

Result: **NOT WORKING only WHITE SCREEN**

### **2.) MCUFRIEND_kbv + ELECHOUSE due shield + ILI9341 screen + DUE**

> I then tried MCUFRIEND_kbv, 
> This is the [MCUFRIEND_kbv](https://github.com/SyberStark/Due-elechouse-tft-Shield-V2.1-TFT_320QDT_9341-Boiana_EQ.git) after I made changes

_Following are the changes made in MCUFRIEND_kbv_

1. Uncommented only #define USE_ELECHOUSE_DUE_16BIT_SHIELD in mcufriend_special.h and Changed Pins According to ELECHOUSE DUE ADAPTER

```
// only define one "USE_XXX" macro at any time

//#define USE_MEGA_8BIT_PROTOSHIELD
//#define USE_MEGA_8BIT_SHIELD      // 4.7sec Mega2560 Shield
//#define USE_MEGA_8BIT_PORTC_SHIELD // 4.7sec Mega2560 Shield
//#define USE_MEGA_16BIT_SHIELD     // 2.14sec Mega2560 Shield 
//#define USE_BLD_BST_MEGA32U4
//#define USE_BLD_BST_MEGA2560      // 12.23sec Uno Shield (17.38s C)
//#define USE_BLD_BST_MEGA4809      // 5.43sec XPRO-Adapter (7.09s C)
//#define USE_DUE_8BIT_PROTOSHIELD
//#define USE_DUE_16BIT_SHIELD        //RD on PA15 (D24) 
//#define USE_BOBCACHELOT_TEENSY
//#define USE_OPENSMART_SHIELD_PINOUT_UNO
//#define USE_OPENSMART_SHIELD_PINOUT_MEGA
//#define USE_OPENSMART_SHIELD_PINOUT_DUE //thanks Michel53
#define USE_ELECHOUSE_DUE_16BIT_SHIELD    //Untested yet
//#define USE_MY_BLUEPILL
//#define USE_ADIGITALEU_TEENSY
//#define USE_MIKROELEKTRONIKA
//#define USE_XPRO_MEGA4809
//#define USE_MY_PICO
//#define USE_CURIOSITY_AVR128DA48
//#define USE_CURIOSITY_AVR128DB48
```
```
#elif defined(__SAM3X8E__) && defined(USE_ELECHOUSE_DUE_16BIT_SHIELD)  //ELECHOUSE_DUE shield on DUE
#warning USE_ELECHOUSE_DUE_16BIT_SHIELD
#define USES_16BIT_BUS
// configure macros for the control pins
#define RD_PORT PIOA
[... 66 lines cut ...]
#define SUPPORT_68140             //RM68140 +52 bytes defaults to PIXFMT=0x55
//#define SUPPORT_7735
#define SUPPORT_7781              //ST7781 +172 bytes
//#define SUPPORT_8230              //UC8230 +118 bytes
//#define SUPPORT_8347D             //HX8347-D, HX8347-G, HX8347-I, HX8367-A +520 bytes, 0.27s
//#define SUPPORT_8347A             //HX8347-A +500 bytes, 0.27s
//#define SUPPORT_8352A             //HX8352A +486 bytes, 0.27s
//#define SUPPORT_8352B             //HX8352B
//#define SUPPORT_8357D_GAMMA       //monster 34 byte 
//#define SUPPORT_9163              //
//#define SUPPORT_9225              //ILI9225-B, ILI9225-G ID=0x9225, ID=0x9226, ID=0x6813 +380 bytes
#define SUPPORT_9320              //ID=0x0001, R61505, SPFD5408, ILI9320
#define SUPPORT_9325              //RM68090, ILI9325, ILI9328, ILI9331, ILI9335 
#define SUPPORT_9326_5420         //ILI9326, SPFD5420 +246 bytes
#define SUPPORT_9342              //costs +114 bytes
//#define SUPPORT_9806              //UNTESTED
#define SUPPORT_9488_555          //costs +230 bytes, 0.03s / 0.19s
#define SUPPORT_B509_7793         //R61509, ST7793 +244 bytes
//#define OFFSET_9327 32            //costs about 103 bytes, 0.08s
```

After making these changes, I ran Arduino IDE \File \examples\MCUFRIEND_kbv\ [graphictest_kbv](https://github.com/SyberStark/Due-elechouse-tft-Shield-V2.1-TFT_320QDT_9341-Boiana_EQ/blob/145e2794e7b091097aa7ff7131bc873135cb9c26/examples/graphictest_kbv/graphictest_kbv.ino)
and _"Didnt change anything in graphictest_kbv....just uploaded it in Arduino DUE the way it is"_

Result: **NOT WORKING only WHITE SCREEN**

prenticedavid, 2022-08-29:
_After making these changes, I ran Arduino IDE \File \examples\UTFT\Arduino(ARM)+Teensy\UTFT_Demo_320x240_

That is ominous. Why do you say Teensy when you have a Due?

I would expect UTFT examples to work with EHOUSE_DUE_SHIELD
But it is always wise to re-install a virgin copy of UTFT.  And then edit for EHOUSE_DUE_SHIELD

I would check your EHOUSE adapter i.e. for RD=24, RS=25, WR=26, CS=27, RST=28 according to v1.0
DB0..DB7 on PIOC2..PIOC9,  DB8..DB15 on PIOC12..PIOC19
Download "Pin Mapping.pdf" from ElecHouse
It is not unknown for a new version to change the routeing.  

I would concentrate on virgin UTFT first. 

Then set appropriate defines in LCD_ID_readreg.ino and  post the report.
If that reads correctly,  install virgin MCUFRIEND_kbv.  edit `USE_SPECIAL, USE_ELECHOUSE_DUE_16BIT_SHIELD`

David.

prenticedavid, 2022-08-30:
Oops.  I wrote  RD=24, RS=25, WR=26, CS=27, RST=28 in the earlier message.

When I looked closely at "Pin Mapping.pdf" from ElecHouse it has RD=24, RS=25, WR=26, CS=31 (PA7), RST=33 (PC1)
e.g. you would edit LCD_ID_readreg.ino for EHOUSE on the Due
```
#define LCD_RST 33 //A4
#define LCD_CS 31 //A3
#define LCD_RS 22 //A2
#define LCD_WR 23 //A1
#define LCD_RD 24 //A0

#define LCD_D0 34 //8
#define LCD_D1 35 //9
#define LCD_D2 36 //2
#define LCD_D3 37 //3
#define LCD_D4 38 //4
#define LCD_D5 39 //5
#define LCD_D6 40 //6
#define LCD_D7 41 //7
```

LCD_ID_readreg.ino sketch uses no libraries at all.   Only digitalRead() and digitalWrite()  

David.

SyberStark, 2022-08-31:
Dear David. Thank u for your reply.

### **1.) New UTFT**
Like u suggested, I replaced the Old UTFT with the [**New UTFT**  ](http://www.rinkydinkelectronics.com/download.php?f=UTFT.zip) 

But, this (and most other) UTFT Libraries have examples of: ARM + Teensy, AVR, Chipkit, TI Launchpads(Energia)
`Arduino IDE \File \examples\UTFT\ Arduino(ARM)+Teensy\UTFT_Demo_320x240`

And ofcourse none of those examples work with my Hardware. 

Can U suggest me any other UTFT Example compatible for ARM+DUE ?
Or maybe a New UTFT Library ?

***Additional Info***
I came across this [UTFT](https://github.com/SyberStark/UTFT-_DUE_9341-16/tree/main/UTFT/examples) which has only 
ARM written in example `Arduino IDE \File \examples\UTFT\Arduino(ARM)\UTFT_Demo_320x240`
But, It doesn't have drivers ILI9341_16 and EHOUSE compatiablity.

> Note: The [EHOUSE ADAPTER](https://www.elechouse.com/elechouse/images/product/3.2TFT%20shield/Pin%20Mapping.pdf), I mentioned above doesn't it have 
> _RD=24, RS=22, WR=23, CS=31, RST=33_ as this? 
> as Mine is ELECHOUSE tft Shield V2.1, and it has the exact pinouts as mentioned. 

 ### **2.) LCD_ID_readreg.ino**
As for LCD_ID_readreg.ino, Do u want me to run the one present in 
`Arduino IDE \File \examples\MCUFRIEND_kbv\LCD_ID_readreg.ino` ???

prenticedavid, 2022-08-31:
If the UTFT examples are for ARM,  they should work.
You should use the proper UTFT i.e. from RinkyDink i.e. v2.83

Your other link goes to an obsolete v2.70

If your EHOUSE v2.1 adapter has the same pinout as v1.0 there must be a reason for changing the major version number.   Please compare shape of pcb and components.

Yes,  run any LCD_ID_readreg.ino from MCUFRIEND_kbv but you need to set the appropriate defines like my post..

Your earlier post said that the display works on the Mega2560 Adapter Shield.
Why do you want to use the Due ?
What country are you in?

David.

SyberStark, 2022-09-01:
Dear David. Thank u for your reply.

> 
> I need to use DUE cause i want to RUN Boiana_EQ code which, only seems to be compatible with DUE as it is fast. 
> I m OK to use Mega2560 as well if the Boiana_EQ code works on it. Currently its not. 
> 
> India

### **1.) UTFT**

I couldn't figure out where the version is mentioned in the UTFT Library folder.
_(documentation/version.txt only explains the changes in version, doesn't tells which version of UTFT we are using)_

But I downloaded this [UTFT](http://www.rinkydinkelectronics.com/library.php?id=51) from rinky-dinky. Which appears to be the latest version 2.83 

Uncommented the line in HW_ARM_defines.h 
and used this [ARM example](https://github.com/SyberStark/UTFT-_DUE_9341-16/tree/main/UTFT/examples/Arduino%20(ARM)/UTFT_Demo_320x240) from a different UTFT
(as the UTFT I downloaded from rinky-dinky,  have examples of: ARM + Teensy, AVR, Chipkit, TI Launchpads(Energia))  

**Result: NOT WORKING only white screen**

I did check the [ELECHOUSE 1.0V](https://camo.githubusercontent.com/a000879eee164339baeaef70cc562c6a51ccf29fa941e2273818003dd81934a2/687474703a2f2f7777772e656c6563686f7573652e636f6d2f656c6563686f7573652f696d616765732f70726f647563742f332e32544654253230736869656c642f7466742d736869656c642d352e6a7067) and [ELECHOUSE 2.0V](https://www.elechouse.com/elechouse/images/product/5.0-TFT/5.0-TFT-1.jpg), Looks like 1.0V Misses 2 resistors on top when compared with [my2.1V](https://user-images.githubusercontent.com/58384591/186730669-08b5a3a9-cff3-4630-b43d-6393a3a03aa3.jpeg). Beside the shape of PCB looks the same. 
I couldn't find the Pin Mapping for 1.0V. 

> Although In [This Fourm](https://forum.arduino.cc/t/getting-a-ssd1963-chipset-tft-on-elechouse-v2-1-shield-to-work-with-adafruit_gfx/469641/6), 'dillonc' uses 2.0V with MCUFRIEND_kbv and UTFT(post#10) claiming both of them works.  
> (He uses same [Pin Mapping](https://www.elechouse.com/elechouse/images/product/3.2TFT%20shield/Pin%20Mapping.pdf) as Mine)
> Hence I think the PinMapping of 2.1V , 2.0V , 1.0V are same.

### **2.) LCD_ID_readreg.ino**

Code: After Changing mcufriend_special.h,mcufriend_shield.h
```
#define LCD_RST 33
#define LCD_CS 31
#define LCD_RS 22
#define LCD_WR 23
#define LCD_RD 24

#define LCD_D0 34
#define LCD_D1 35
#define LCD_D2 36
#define LCD_D3 37
#define LCD_D4 38
#define LCD_D5 39
#define LCD_D6 40
#define LCD_D7 41

void setup()
{
    Serial.begin(9600);
    while (!Serial) ;
    Serial.println("Read Registers on MCUFRIEND UNO shield");
    Serial.println("controllers either read as single 16-bit");
    Serial.println("e.g. the ID is at readReg(0)");
    Serial.println("or as a sequence of 8-bit values");
    Serial.println("in special locations (first is dummy)");
    Serial.println("");
    lcdInit();
    lcdReset();      //ensures that controller is in default state
//    for (uint16_t i = 0; i < 256; i++) readReg(i, 7, "f.k"); 
    readReg(0x00, 2, "ID: ILI9320, ILI9325, ILI9335, ...");
    readReg(0x04, 4, "Manufacturer ID");
    readReg(0x09, 5, "Status Register");
    readReg(0x0A, 2, "Get Power Mode");
    readReg(0x0C, 2, "Get Pixel Format");
    readReg(0x61, 2, "RDID1 HX8347-G");
    readReg(0x62, 2, "RDID2 HX8347-G");
    readReg(0x63, 2, "RDID3 HX8347-G");
    readReg(0x64, 2, "RDID1 HX8347-A");
    readReg(0x65, 2, "RDID2 HX8347-A");
[... 240 lines cut ...]
___________________________________________________________________________________________________________________________

with your PinMaps
```
// adapted from LCD_ID_Reader from http://misc.ws/lcd_information
// controllers either read as 16-bit or as a sequence of 8-bit values

//-- Arduino UNO or Mega 2560 Plugged as shield
#define LCD_RST 33
#define LCD_CS 31
#define LCD_RS 25
#define LCD_WR 26
#define LCD_RD 24

#define LCD_D0 34
#define LCD_D1 35
#define LCD_D2 36
#define LCD_D3 37
#define LCD_D4 38
#define LCD_D5 39
#define LCD_D6 40
#define LCD_D7 41
```

Result : Same As above

prenticedavid, 2022-09-01:
Please do NOT paste large swathes of code when you are simply using a public example sketch.
Make sure that you start with the virgin code.   Then just paste the changed lines e.g like I did https://github.com/prenticedavid/MCUFRIEND_kbv/issues/213#issuecomment-1232259257

And if I tell you that LCD_ID_readreg does not use any libraries,  please don't say that you have edited libraries.

I don't believe that  LCD_ID_readreg will ever crash your Due.
It appears that dillonc from your link had success with EHOUSE and SSD1963 in 2017.
You can try to contact him.

Since you are in India it is not practical to mail your Adapter.
If you can find an online shop that sells this EHOUSE adapter I will buy one for myself.
You can certainly buy the CTE Due Adapter.   I possess one.

David.

SyberStark, 2022-09-01:
Dear David. Thank for your reply.

By virgin code, you meant ==> A fresh code with no changes made to it right? If yes. 
Then I Did exactly like that, I took a New Code for LCD_ID_readreg.ino and then changed the required Pins accordingly.
Result is the same as posted above. 

I expected at least LCD_ID_readreg.ino to work with my Hardware (Due + Elechouse Shield + ILI9341)
What according to you could be the possible fault?

EHOUSE shield for Due can be bought from [here.](https://www.emartee.com/product/42228/3.2%22%20and%205%22%20TFT%20Shield%20for%20Arduino%20Due)

Can you post the Image or Buying Link of your CTE Due Adapter?

prenticedavid, 2022-09-01:
Yes.  Use LCD_ID_readreg.ino from GitHub or fresh library installation.  Edit defines for EHOUSE Adapter.

Thanks for the link.  I have ordered the EHOUSE adapter.

Re CTE:  Google for "TFT Adapter Shield Due".   e.g. https://www.ebay.co.uk/itm/144189566145

SyberStark, 2022-09-02:
Dear David. Thank u for your reply.

I downloaded a fresh [MCUFRIEND_kbv](https://github.com/prenticedavid/MCUFRIEND_kbv)

With due Same result.

With Mega2560 + [Mega Shield](https://user-images.githubusercontent.com/58384591/186731304-121efcec-cb17-493d-bc35-323fdfaeec28.jpeg) + ILI9341, I Made these changes :
```
#define LCD_RST 41
#define LCD_CS 40
#define LCD_RS 38
#define LCD_WR 39
#define LCD_RD 43   //(Not 100% sure about RD pin value)

#define LCD_D0 22
#define LCD_D1 23
#define LCD_D2 24
#define LCD_D3 25
#define LCD_D4 26
#define LCD_D5 27
#define LCD_D6 28
#define LCD_D7 29
```

Result:
```
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 04 04 04 04	Manufacturer ID
reg(0x0009) 09 09 09 09 09	Status Register
reg(0x000A) 0A 0A	Get Power Mode
reg(0x000C) 0C 0C	Get Pixel Format
reg(0x0061) 61 61	RDID1 HX8347-G
reg(0x0062) 62 62	RDID2 HX8347-G
reg(0x0063) 63 63	RDID3 HX8347-G
reg(0x0064) 64 64	RDID1 HX8347-A
reg(0x0065) 65 65	RDID2 HX8347-A
reg(0x0066) 66 66	RDID3 HX8347-A
reg(0x0067) 67 67	RDID Himax HX8347-A
reg(0x0070) 70 70	Panel Himax HX8347-A
reg(0x00A1) A1 A1 A1 A1 A1	RD_DDB SSD1963
reg(0x00B0) B0 B0	RGB Interface Signal Control
reg(0x00B4) B4 B4	Inversion Control
reg(0x00B6) B6 B6 B6 B6 B6	Display Control
reg(0x00B7) B7 B7	Entry Mode Set
reg(0x00BF) BF BF BF BF BF BF	ILI9481, HX8357-B
reg(0x00C0) C0 C0 C0 C0 C0 C0 C0 C0 C0	Panel Control
reg(0x00C8) C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8	GAMMA
reg(0x00CC) CC CC	Panel Control
reg(0x00D0) D0 D0 D0	Power Control
reg(0x00D2) D2 D2 D2 D2 D2	NVM Read
reg(0x00D3) D3 D3 D3 D3	ILI9341, ILI9488
reg(0x00D4) D4 D4 D4 D4	Novatek ID
reg(0x00DA) DA DA	RDID1
reg(0x00DB) DB DB	RDID2
reg(0x00DC) DC DC	RDID3
reg(0x00E0) E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0	GAMMA-P
reg(0x00E1) E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1	GAMMA-N
reg(0x00EF) EF EF EF EF EF EF	ILI9327
reg(0x00F2) F2 F2 F2 F2 F2 F2 F2 F2 F2 F2 F2 F2	Adjust Control 2
reg(0x00F6) F6 F6 F6 F6	Interface Control

```
I think its not even reading the ID with My working Hardware.

> Beside Thankx for the link I have ordered CTE, but It might take around a month to receive

prenticedavid, 2022-09-02:
The Mega2560 Adapter is **write-only**.   There is no possibility of reading ID etc with MCUFRIEND_kbv.
If you look at your **Mega2560** report you would notice that the "reg" is just echoed i.e. RD is not functional

Your ElecHouse Due Adapter has a RD pin.   As far as I know LCD_ID_readreg should work out of the box.  i.e. with the EHOUSE defines.

The CTE Due Adapter does not have a RD pin.   But it is a very simple hardware mod to connect the RD pin.   I have done the hardware mod on my CTE Due Adapter.   This makes SSD1963 40-pin screen work 100% with MCUFRIEND-kbv.   I can run LCD_ID_readreg on my SSD1963 screen (with CTE adapter)

The whole design idea behind MCUFRIEND_kbv was to read the ID and run with many different TFT controllers.

You can run MCUFRIEND_kbv on write-only screens but you must set the correct ID manually.   e.g. `tft.begin(0x9341)`
Obviously the read methods do **not** work.

I have no idea or interest in Boiana_EQ project.   Does it need to read any registers?
It should work fine on the Mega2560.   The Mega2560 should be fast enough with your White Adapter Shield.

SyberStark, 2022-09-14:
Dear David. Thank u for your reply.

My screen is working with:
DUE + Mega2560 + ILI9341 + graphictest_kbv `(I set ID manually tft.begin(0x9341))`
(Changed mcufriend_shield.h, mcufriend_special.h accordingly)

Yet a white screen with 
DUE + ELECHOUSE adapter + ILI9341 + graphictest_kbv  `(I set ID manually tft.begin(0x9341))`
(Changed mcufriend_shield.h, mcufriend_special.h accordingly)
Do u think it has something to do with **pin_shield_1.h** ? As mine are these by default: 
```
#elif defined(__SAM3X8E__)
  #include <sam3xa.h>
  #define D0_PORT PIOA
  #define D0_PIN  8   
  #define D1_PORT PIOA
  #define D1_PIN  9
  #define D2_PORT PIOB
  #define D2_PIN  25
  #define D3_PORT PIOC
  #define D3_PIN  28
  #define D4_PORT PIOC   //also PA29
  #define D4_PIN  26
  #define D5_PORT PIOC
  #define D5_PIN  25
  #define D6_PORT PIOC
  #define D6_PIN  24
  #define D7_PORT PIOC
  #define D7_PIN  23
  #define D8_PORT PIOC
  #define D8_PIN  22
  #define D9_PORT PIOC
  #define D9_PIN  21
  #define D10_PORT PIOC  //also PA28
  #define D10_PIN  29
  #define D11_PORT PIOD
  #define D11_PIN  7
  #define D12_PORT PIOD
  #define D12_PIN  8
  #define D13_PORT PIOB
  #define D13_PIN  27
  #define A0_PORT PIOA
  #define A0_PIN  16
  #define A1_PORT PIOA
  #define A1_PIN  24
  #define A2_PORT PIOA
  #define A2_PIN  23
  #define A3_PORT PIOA
  #define A3_PIN  22
  #define A4_PORT PIOA
  #define A4_PIN  6
  #define A5_PORT PIOA
  #define A5_PIN  4
```

Im still waiting for CTE Due Adapter, have you received ELECHOUSE Adapter?

prenticedavid, 2022-09-14:
I don't trust Emartee.com.  The "Order 10114" just says "Processing".   I suspect that this just means "out of stock" and we are not going to admit it.

pin_shield_1.h is not used by Arduino.   Only by Keil, Rowley, ...

_"My screen is working with:
DUE + Mega2560 + ILI9341 + graphictest_kbv (I set ID manually tft.begin(0x9341))"_
What is that supposed to mean?

The White Mega2560 Adapter shown in your Original Post should plug into MEGA2560 board.
It should work just fine with UTFT.`   UTFT myGLCD(ILI9341_16,38,39,40,41);`
And with MCUFRIEND_kbv if you USE_SPECIAL, USE_MEGA_16BIT_SHIELD, tft.begin(0x9341)

The White Mega2560 Adapter shown in your Original Post should plug into DUE board.
It should work just fine with UTFT.   `   UTFT myGLCD(ILI9341_16,38,39,40,41);`
And with MCUFRIEND_kbv if you USE_SPECIAL, USE_MEGA_16BIT_SHIELD, tft.begin(0x9341)

Seriously.  You can test the EHOUSE Adapter by running LCD_ID_readreg.ino with the the EHOUSE defines.  i.e.

```
#define LCD_RST 33
#define LCD_CS 31
#define LCD_RS 25
#define LCD_WR 26
#define LCD_RD 24

#define LCD_D0 34
#define LCD_D1 35
#define LCD_D2 36
#define LCD_D3 37
#define LCD_D4 38
#define LCD_D5 39
#define LCD_D6 40
#define LCD_D7 41
```

This should give meaningful results e.g.
```
reg(0x00D3) 00 93 41 00	ILI9341, ILI9488
```
If it does not then the EHOUSE Adapter is wrong (or I am wrong)

SyberStark, 2022-09-16:
Dear David. Thank u for your reply.

_"The White Mega2560 Adapter shown in your Original Post should plug into DUE board.
It should work just fine with UTFT. UTFT myGLCD(ILI9341_16,38,39,40,41);
And with MCUFRIEND_kbv if you USE_SPECIAL, USE_MEGA_16BIT_SHIELD, tft.begin(0x9341)"_
Yes, I meant with these hardwares & configuration screen is working. 

I Tried, EHOUSE Adapter + DUE + ILI9341 + LCD_ID_readreg.ino
Same result as [Above Post](https://github.com/prenticedavid/MCUFRIEND_kbv/issues/213#issuecomment-1234224958)

I am going to order a new EHOUSE Adapter. Even though its quite hard to digest that the EHOUSE Adapter is defected.
I have checked the continuity(with a multimeter) of each pin of EHOUSE Adapter with respect to ILI9341, and they all seems fine...**NO** sign of the adapter being physically damaged too.

prenticedavid, 2022-09-17:
You are never going to follow my suggestion to _You can test the EHOUSE Adapter by running LCD_ID_readreg.ino with the the EHOUSE defines_

So there is no point in me wasting my time.

SyberStark, 2022-09-17:
Dear David, Thankx for your reply.

Im sorry...I am following all your suggestions. Maybe I miss understood. 
I interpreted this _"You can test the EHOUSE Adapter by running LCD_ID_readreg.ino with the the EHOUSE defines"_ as below:

I connected "DUE" with "ILI9341" with "EHOUSE ADAPTER" and ran LCD_ID_readreg.ino (along with EHOUSE defines u mentioned i.e.)
```
#define LCD_RST 33
#define LCD_CS 31
#define LCD_RS 25
#define LCD_WR 26
#define LCD_RD 24

#define LCD_D0 34
#define LCD_D1 35
#define LCD_D2 36
#define LCD_D3 37
#define LCD_D4 38
#define LCD_D5 39
#define LCD_D6 40
#define LCD_D7 41
```

Result I got is:
```
ReadEND UNO shield
controllers either read as single 16-bit⸮Read Registers on MCUFRIEND UNO shield
controllers either re on MCUFRIEND UNO shield
controllers either read as single 16-bit⸮Read Registers on MCUFRIEND UNO shield
controllers either re on MCUFRIEND UNO shield
controllers either read as single 16-bit⸮Read Registers on MCUFRIEND UNO shield
controllers either re⸮⸮as single 1⸮Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit⸮Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit⸮Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit⸮Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit⸮Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit⸮Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit⸮Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit⸮Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit⸮Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit⸮Read Registers on MCUFRIEND UNO shield
```

May you please elaborate on what am I supposed to do?

prenticedavid, 2022-09-17:
I don't believe you.  Please plug the Display + EHOUSE + Due again.
I don't see how LCD_ID_readreg.ino could reset the Arduino

SyberStark, 2022-09-17:
Dear David, Thankx for your reply.

I Reconnected Everything and tried it out again. Here is the [video](https://youtu.be/sKUdfsQyyPY):

Let me know where I am wrong.

prenticedavid, 2022-09-17:
Your video shows a LDO Voltage Regulator chip and a 3x1 pin header.   The pin header has a black jumper-link on 2 of the 3x1 pins.
Incidentally pictures in https://github.com/elechouse/tft_shield do not have this.

It should be pretty easy to trace the wiring of these components.   Either with a bright light or a DMM.
Likewise **you** can trace the "logic" pins on the header to compare with the "pin mapping.pdf".

When you trace the LDO connections and 3x1 connections I expect you to write them down properly or draw the accurate schematic.   I don't expect to see the "pin mapping.pdf" in the schematic.   Just a YES or NO to whether you have checked all routes.

I doubt if I will ever receive an Adapter from EMartee.    But if I did it would take 10 minutes to check each pin with a DMM and 5minutes to draw the LDO and 3x1 schematic.

SyberStark, 2022-09-22:
Dear David, Thankx for your reply.

Its a 3.3V voltage regulator, This DUE adapter can provide 3.3V or 5V to the Screen. 
I have connected the black jumper-link to 5V (as 3.3V gives dull backlight for my ILI9341)

 
YES ALL THE ROUTES are CHECKED. 
I traced the wiring of these components with DMM. The pin mapping came out to be exactly as the one posted [above.](https://www.elechouse.com/elechouse/images/product/3.2TFT%20shield/Pin%20Mapping.pdf) 
[image]

Checking 3.3V Regulator with DMM got me this:
[image]

 
You'll receive the Adapter from Ematee, don't worry. When I ordered it the status was showing 'received'  even though I didn't receive it.
But after 1 month I received the Due adapter from them.

prenticedavid, 2022-09-22:
Ah-ha.  So you bought your ElecHouse Adapter from Emartee.com
They have never replied to emails.
Have you received your CTE/Sainsmart Adapter yet ?

Have you tried the White Adapter on the Due ?
It should work with the same UTFT constructor that you used on the MEGA2560.
Remember to `#undef EHOUSE_DUE_SHIELD` and `#undef CTE_DUE_SHIELD`

It will not work with LCD_ID_readreg.ino because it is **write-only**.
It should work with MCUFRIEND_kbv with `USE_SPECIAL, USE_MEGA_16BIT_SHIELD, tft.begin(0x9341)`

Anyway,  it seems logical for the chip on the Adapter to be a 3.3V regulator.   The SMD components look like Capacitors and not resistors.