prenticedavid/MCUFRIEND_kbv #228: 3.5 mcufriend tft only (white screen)

Osama-Abd-El-Mohsen, 2023-02-20:
hi i bought 3.5 mcufriend tft but i can not deal with it all i have only (white screen)

the store link : [LCD Touch Module 3.5 Inch Tft - Electra](https://electra.store/product/lcd-touch-module-3-5-inch-tft/)
the product pics :
[image]

i tried all libraries utft and mcufriend etc but nothing at all
i am using the tft with arduino mega
when i used mcufriend lib example grahic test_kbv i got
```

Serial took 0ms to start
ID = 0x404
```
and some times `ID = 0x808`

i changed mcufriend_shield.h and added

`#define USE_SPECIAL`

and changed mcufriend_spechial.h and un commented
`#define USE_MEGA_8BIT_SHIELD
`
then tried the example again i got

```
Serial took 0ms to start
ID = 0xD3D3
```
pls help and thx

Osama-Abd-El-Mohsen, 2023-02-20:
and that wht i get from example LCD_ID_readreg
```
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 04 04	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 04 04 04 04	Manufacturer ID
reg(0x0009) 05 05 05 05 05	Status Register
reg(0x000A) 06 06	Get Power Mode
reg(0x000C) 04 04	Get Pixel Format
reg(0x0061) 65 65	RDID1 HX8347-G
reg(0x0062) 66 66	RDID2 HX8347-G
reg(0x0063) 67 67	RDID3 HX8347-G
reg(0x0064) 64 64	RDID1 HX8347-A
reg(0x0065) 65 65	RDID2 HX8347-A
reg(0x0066) 66 66	RDID3 HX8347-A
reg(0x0067) 67 67	RDID Himax HX8347-A
reg(0x0070) 74 74	Panel Himax HX8347-A
reg(0x00A1) A5 A5 A5 A5 A5	RD_DDB SSD1963
reg(0x00B0) B4 B4	RGB Interface Signal Control
reg(0x00B4) B4 B4	Inversion Control
reg(0x00B6) B6 B6 B6 B6 B6	Display Control
reg(0x00B7) B7 B7	Entry Mode Set
reg(0x00BF) BF BF BF BF BF BF	ILI9481, HX8357-B
reg(0x00C0) 44 44 44 44 44 44 44 44 44	Panel Control
reg(0x00C8) 44 44 44 44 44 44 44 44 44 44 44 44 44	GAMMA
reg(0x00CC) 44 44	Panel Control
reg(0x00D0) 54 54 54	Power Control
reg(0x00D2) 56 56 56 56 56	NVM Read
reg(0x00D3) 57 57 57 57	ILI9341, ILI9488
reg(0x00D4) 54 54 54 54	Novatek ID
reg(0x00DA) 56 56	RDID1
reg(0x00DB) 57 57	RDID2
reg(0x00DC) 54 54	RDID3
reg(0x00E0) E4 E4 E4 E4 E4 E4 E4 E4 E4 E4 E4 E4 E4 E4 E4 E4	GAMMA-P
reg(0x00E1) E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5 E5	GAMMA-N
reg(0x00EF) EF EF EF EF EF EF	ILI9327
reg(0x00F2) F6 F6 F6 F6 F6 F6 F6 F6 F6 F6 F6 F6	Adjust Control 2
reg(0x00F6) F6 F6 F6 F6	Interface Control
```

brightproject, 2023-03-20:
`reg(0x00D3) 57 57 57 57	ILI9341, ILI9488`
You have not correctly connected the display to the microcontroller.

closed 2023-03-20 by Osama-Abd-El-Mohsen