prenticedavid/MCUFRIEND_kbv #232: 'GFXglyph' does not name a type

FredLeliveld, 2023-04-07:
I get this error ('GFXglyph' does not name a type) when I run the example : **Font_simple** witch comes with version: MCUFRIEND_kbv-3.0.0-Release. 
To prevent a former Error ('GFXfont' does not name a type) I put "FreeDefaultFonts.h" in the same map as "Font_simple.ino"
But now I get the error: 'GFXglyph' does not name a type'.
I have no idee what I am doing wrong, because I use the Font_simple.ino as it comes with the package.
Can someone please help me to get the program work so I can use freeFonts in my project.
Fred (Holland)

haoppia, 2024-08-10:
i have the same problem when i compiled the e-paper weather

Andy2No, 2024-08-10:
@haoppia Which chip are you targeting, which core or boards support package are you using and which version does it show in the Arduino Board Manager?  Are you using the Arduino IDE or Platform IO?

haoppia, 2024-08-10:
@Andy2No I use the esp32-wroom-32e board,i'm totally a freshman of the field using the Arduino IDE. I downloaded the e-paper-weather-epd from github of imarzen,and encountered to this problem,hope for your solution,thank u

Andy2No, 2024-08-10:
I see.  For the ESP32, the boards support version usually matters.  When they released 3.0.x versions, it broke compatibility for software written for 2.0.x.   The release of 2.0.x broke support for software written for 1.0.x too.

I wasn't able to get anything compiled for FabGL until I found it used 2.0.17 or earlier.  I'd installed the latest ESP32 boards support, which was 3.0.2 at the time.

So, the thing to do it to try to find out which boards support version was used for the example you're trying to find.  Failing that, try uninstalling the one you have, in the Boards Manager of the Arduino IDE (use Tools->Board->Boards manager, then put ESP32 in the search box), then install v2.0.17 if you had 3.0.x installed.

You'll probably see more than one result for ESP32 in the Boards manager, but you want the one written by Espressif Systems.

This becomes a problem if you need to use ESP32 software written for different ESP32 boards support versions.  Apparently, Platform IO can allow you to specify which one to use, but I've never tried that.

haoppia, 2024-08-10:
Thank u so much, i will try that for different version

 ----------Reply to Message----------

  

 

 
 
So, the thing to do it to try to find out which boards support version was used for the example you're trying to find.  Failing that, try uninstalling the one you have, in the Boards Manager of the Arduino IDE (use Tools-&gt;Board-&gt;Boards manager, then put ESP32 in the search box), then install v2.0.17 if you had 3.0.x installed.

Andy2No, 2024-08-10:
Also, do you have the Adafruit graphics library installed?  That's where the GFXglyph definition comes from.

This issue for another TFT LCD library might be relevant, which is also based on Adafruit_GFX:

https://github.com/Bodmer/TFT_eSPI/issues/1688

haoppia, 2024-08-10:
Hasn't yet ,I only installed the Adafruit_ GFXglyph library

 ----------Reply to Message----------

  

 

 
 
Bodmer/TFT_eSPI#1688

Andy2No, 2024-08-10:
Ah.  Okay.  Maybe work through the instructions on the Code tab /  main page: https://github.com/prenticedavid/MCUFRIEND_kbv

Perhaps start with the examples and find ones that work - they might not all work straight away but finding ones that do will give you more confidence that it's going in the right direction.

haoppia, 2024-08-10:
Okay. Really thank u for your help. I will try it then.

 ----------Reply to Message----------

Andy2No, 2024-08-10:
You're welcome.