prenticedavid/MCUFRIEND_kbv #239: Won't compile on Uno R4 Wifi

git-eri, 2023-07-21:
Tried compiling some examples for Arduino R4 Wifi but it errors out. Same setup working for standart Uno. Any help on getting this library to work on R4 Wifi is appreciated. It seems likje every example errors out this way.

Here are the errors:

```
...
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:174:5: note: in expansion of macro 'RD_IDLE'
     RD_IDLE;
     ^~~~~~~
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1207:38: error: 'RD_PIN' was not declared in this scope
 #define RD_IDLE    PIN_HIGH(RD_PORT, RD_PIN)
                                      ^
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:174:5: note: in expansion of macro 'RD_IDLE'
     RD_IDLE;
     ^~~~~~~
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1207:38: note: suggested alternative: 'R_PFS'
 #define RD_IDLE    PIN_HIGH(RD_PORT, RD_PIN)
                                      ^
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:174:5: note: in expansion of macro 'RD_IDLE'
     RD_IDLE;
     ^~~~~~~
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1207:20: error: 'PIN_HIGH' was not declared in this scope
 #define RD_IDLE    PIN_HIGH(RD_PORT, RD_PIN)
                    ^
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:174:5: note: in expansion of macro 'RD_IDLE'
     RD_IDLE;
     ^~~~~~~
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1207:20: note: suggested alternative: 'PIN_MISO'
 #define RD_IDLE    PIN_HIGH(RD_PORT, RD_PIN)
                    ^
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:174:5: note: in expansion of macro 'RD_IDLE'
     RD_IDLE;
     ^~~~~~~
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:176:5: error: 'setWriteDir' was not declared in this scope
     setWriteDir();
     ^~~~~~~~~~~
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:176:5: note: suggested alternative: 'setWriteError'
     setWriteDir();
     ^~~~~~~~~~~
     setWriteError
In file included from C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:38:0:
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp: In member function 'int16_t MCUFRIEND_kbv::readGRAM(int16_t, int16_t, uint16_t*, int16_t, int16_t)':
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1215:28: error: 'CS_PORT' was not declared in this scope
 #define CS_ACTIVE  PIN_LOW(CS_PORT, CS_PIN)
                            ^
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:310:9: note: in expansion of macro 'CS_ACTIVE'
         CS_ACTIVE;
         ^~~~~~~~~
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1215:28: note: suggested alternative: 'R_PORT0'
 #define CS_ACTIVE  PIN_LOW(CS_PORT, CS_PIN)
                            ^
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:310:9: note: in expansion of macro 'CS_ACTIVE'
         CS_ACTIVE;
         ^~~~~~~~~
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1215:37: error: 'CS_PIN' was not declared in this scope
 #define CS_ACTIVE  PIN_LOW(CS_PORT, CS_PIN)
                                     ^
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:310:9: note: in expansion of macro 'CS_ACTIVE'
         CS_ACTIVE;
         ^~~~~~~~~
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1215:37: note: suggested alternative: 'SPI'
 #define CS_ACTIVE  PIN_LOW(CS_PORT, CS_PIN)
                                     ^
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:310:9: note: in expansion of macro 'CS_ACTIVE'
         CS_ACTIVE;
         ^~~~~~~~~
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1215:20: error: 'PIN_LOW' was not declared in this scope
 #define CS_ACTIVE  PIN_LOW(CS_PORT, CS_PIN)
                    ^
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:310:9: note: in expansion of macro 'CS_ACTIVE'
[... 611 lines cut ...]
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1213:20: note: suggested alternative: 'PIN_MISO'
 #define CD_DATA    PIN_HIGH(CD_PORT, CD_PIN)
                    ^
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:880:17: note: in expansion of macro 'CD_DATA'
                 CD_DATA;
                 ^~~~~~~
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1216:20: error: 'PIN_HIGH' was not declared in this scope
 #define CS_IDLE    PIN_HIGH(CS_PORT, CS_PIN)
                    ^
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:888:13: note: in expansion of macro 'CS_IDLE'
             CS_IDLE;
             ^~~~~~~
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1216:20: note: suggested alternative: 'PIN_MISO'
 #define CS_IDLE    PIN_HIGH(CS_PORT, CS_PIN)
                    ^
C:\Users\semod\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:888:13: note: in expansion of macro 'CS_IDLE'
             CS_IDLE;
             ^~~~~~~

exit status 1

Compilation error: exit status 1
```

I suspect that there is no support for that type of Arduino Uno. I hope i could get some help with that since i would like to run my ILI9486 display using that Uno R4 Wifi

AdriaBergeAguilar, 2023-08-28:
I have the same problem, and if I put the library in an arduino uno rev3 or lower, it fills up the entire memory (98%) and I have no space left for my programming. Is there a plan to fix this problem?

FrightRisk, 2023-11-03:
It is very unfortunate that Arduino chose to name this board an "Uno" and use R4 which leads people to think it is just the latest version of an Uno after R3. This is causing undue confusion and support issues for developers. This board uses a completely different processor than an Uno and is incompatible with a lot of libraries what would have to be written. It is about as far apart as a PC is to a Mac. You would have to port this yourself or wait for someone to add support for the R4. In addition, it uses some valuable memory for a hardware function, so may have even LESS available memory than a real Uno, especially if you intend to add WiFi support to a sketch. If you need more memory, use a Mega.

benridler, 2023-12-25:
Give this fork, https://github.com/slviajero/MCUFRIEND_kbv/tree/master, a try, @slviajero has forked this repo and updated the definitions to support R4 boards. You'll need to download it manually and add the library as a zip. They have also opened an MR on this repo, which is awaiting reviews https://github.com/prenticedavid/MCUFRIEND_kbv/pull/240.

andreatoninelli, 2024-03-31:
@benridler Thanks for it, works!