prenticedavid/MCUFRIEND_kbv #248: Arduino UNO + ILI9488

felps3104, 2024-01-04:
I'm having trouble with the usage and ID return.
TFT Shield ILI9488, MCUFRIEND 3.10

My aspect_kbv return is `ID = 0x0` and my TFT has a white screen the whole time.
As said somewhere, OPEN-SMART might have a different pinout, so I changed LCD_ID_readnew to
``` C++
//-- Arduino UNO or Mega 2560 Plugged as shield
#define LCD_RST A1 //A1
#define LCD_CS A3
#define LCD_RS A2
#define LCD_WR A1
#define LCD_RD A0

#define LCD_D0 8
#define LCD_D1 9
#define LCD_D2 10 //10
#define LCD_D3 11 //11
#define LCD_D4 4
#define LCD_D5 13 //13
#define LCD_D6 6
#define LCD_D7 7
```

Still got as a return:
```C++
reg(0x00DA) 00 54	RDID1
reg(0x00DB) 00 80	RDID2
reg(0x00DC) 00 66	RDID3
reg(0x00E0) 66 00 07 0C 05 13 09 36 AA 46 09 10 0D 1A 1E 0F	GAMMA-P
reg(0x00E1) 0F 00 20 23 04 10 06 37 56 49 04 0C 0A 33 37 0F	GAMMA-N
reg(0x00EF) 0F 00 00 00 00 00	ILI9327
reg(0x00F2) 00 58 04 12 02 22 22 FF 0A 90 14 88	Adjust Control 2
reg(0x00F6) 88 00 00 00	Interface Control
```

felps3104, 2024-01-05:
Fun fact, pressure calibration is alright...
```C++
TouchScreen.h GFX Calibration
Making all control and bus pins INPUT_PULLUP
Typical 30k Analog pullup with corresponding pin
would read low when digital is written LOW
e.g. reads ~25 for 300R X direction
e.g. reads ~30 for 500R Y direction

Testing : (A1, D7) = 40
Testing : (A2, D6) = 24
ID = 0x9488

cx=579 cy=601 cz=547  X, Y, Pressure
cx=521 cy=417 cz=618  X, Y, Pressure
cx=549 cy=790 cz=479  X, Y, Pressure
cx=500 cy=627 cz=678  X, Y, Pressure
cx=669 cy=462 cz=613  X, Y, Pressure
```

Should I just consider that the screen is damaged?

CommanderGL, 2024-01-07:
No I'm Having the same issue with my screen (which is capacitive) and it seems to be an issue with `MCUFRIEND_kbv` because it works with just `adafruit_gfx`