prenticedavid/MCUFRIEND_kbv #255: 'GPIO' was not declared in this scope  ESP32

4516N41, 2024-06-11:
`In file included from c:\Users\\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:38:
c:\Users\\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h: In function 'void write_8(uint16_t)':
c:\Users\\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1084:5: error: 'GPIO' was not declared in this scope
 1084 |     GPIO.out_w1tc = map_8(0xFF);  //could define once as DMASK
      |     ^~~~
c:\Users\\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h: In function 'uint8_t read_8()':
c:\Users\\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1090:19: error: 'GPIO' was not declared in this scope
 1090 |     return map_32(GPIO.in);
      |                   ^~~~

exit status 1`

I get this error after the "Arduino Release v3.0.1 based on ESP-IDF v5.1.4" update

walid-76, 2024-06-25:
Hello.
iam facing also the same problem and i have the newest version of arduino, were u able to solve the problem ??

cor053, 2024-06-30:
I have got exactly the same problem as well. GPIO doesn't seem to be defined.

In file included from /Users/Documents/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:38:
/Users/Documents/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h: In function 'void write_8(uint16_t)':
/Users/Documents/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h:1084:5: error: 'GPIO' was not declared in this scope
 1084 |     GPIO.out_w1tc = map_8(0xFF);  //could define once as DMASK
      |     ^~~~
/Users/Documents/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h: In function 'uint8_t read_8()':
/Users/Documents/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h:1090:19: error: 'GPIO' was not declared in this scope
 1090 |     return map_32(GPIO.in);
      |                   ^~~~

exit status 1

Compilation error: exit status 1

cor053, 2024-06-30:
I found this thread.
https://github.com/Bodmer/TFT_eSPI/issues/2628
It claims to have solved the problem recently. But what I understand is that is not in any official release yet.
But read it for yourself.

Jakebrake, 2024-11-01:
Same here, Arduino IDE 2.3.3 & ESP-IDF v5.1, on the ol' ESP-WROVER_KIT v4.1

Mohanad75, 2025-02-14:
> Followed the link [@cor053](https://github.com/cor053) posted and implemented the addition of the include for hal/gpio_ll.h and it fixed my issue. In utility/mcufriend_shield.h after line 1025 I added the include, like as follows:
> 
> ```
> 1024	//################################### ESP32 ##############################
> 1025	#elif defined(ESP32)       //regular UNO shield on TTGO D1 R32 (ESP32)
> 1026
> 1027	#include "hal/gpio_ll.h" // GPIO register functions
> 1028
> 1029	#define LCD_RD  2  //LED
> 1030	#define LCD_WR  4
> ```

This fixed my GPIO error "error: 'GPIO' was not declared in this scope "
And just add it like:

#include <Arduino.h>
#include <SPI.h>
#include <Print.h>
#include "spi_dev.h"
#include "hal/gpio_ll.h"

Worked..
 Thank you @ErrorxF00F

Andy2No, 2025-02-15:
@Mohanad75 for reference, which ESP32 boards support package or library version are you using?  E.g. the version chosen in Tools->Board->Board manager in the Arduino IDE?

This sort of error is caused by using a different boards support version than the one the code in this repo was written for, so it's good if people say what they're using.  Generally, the major version numbers are not backwards or forwards compatible - code written for 3.x will not work on 2.x without modification, for example.

kemploe, 2025-07-05:
@Mohanad75 

Hey,

Thanks a lot for the solution.

You saved mya day!

~ kemploe ~